Atmosphere-NX/Atmosphere #243: white screen after CFW launch

fculpo, 2018-10-24:
In hekate, when launching Atmosphere 0.7.0 to 0.7.3, I only get a white screen just after the end loading kips.

I got all setup files from SDSetup Default (homebrew only)

SciresM, 2018-10-24:
This occurs for all Atmosphere versions?

Could you delete the atmosphere folder on your SD card, and try using a fresh copy of the latest release? Does the issue occur when launching via fusee-primary and not hekate?

fculpo, 2018-10-25:
Switch 6.0.1
Atmosphere 0.7.3
Booted via fusee-primary

Result: ok BUT hbmenu does not find switch folder

fculpo, 2018-10-25:
Archive bit cleared
Now works fine.

I will try to add more sdfiles and see if i can reproduce the white screen.

closed 2018-10-25 by fculpo