Atmosphere-NX/Atmosphere #246: Console does not shut down properly.

maciak, 2018-10-25:
## Bug Report

### What's the issue you encountered?
When trying to shut the console down it seems turned off but joy-cons are still in sync and you cannot enter RCM.
### How can the issue be reproduced?
Just hold power for few sec. and turn off from power menu.
### Crash report?

(If a crash report was created under /atmosphere/crash_reports/, please upload it to 
[gist](https://gist.github.com/) and paste the link here.)

### Environment?

- What bootloader (fusee, hekate, etc) was Atmosphère launched by: Fusee (But happens on any other also)
- Official release or unofficial build: Official 0.7.3
- Do you have additional kips you're loading: No
- Additional info about your environment: FW 5.1 updated through normal software update, no AutoRCM

### Additional context?
Consolle seems to be turned off as screen is black but it's still on and joycons are syncing and I can still controle volume for example. Volume level will change after next reboot.

aroooo, 2018-10-29:
Got that issue too.

ThatNerdyPikachu, 2018-10-30:
.

maciak, 2018-11-26:
Hi,
Would maybe this help with this issue?
https://github.com/Reisyukaku/ReiNX/commit/74695c87bbd57075318c2538a6f84dfc18b51f45

closed 2019-05-23 by SciresM