Atmosphere-NX/Atmosphere #247: Black ribbon hbloader

jul2003, 2018-10-28:
## Bug Report

### What's the issue you encountered?

When launching hbloader from a game app the nintendo black ribbon is displayed so screen is reduced

### How can the issue be reproduced?

Put hbmenu.nro to the root of the sd , hbl.nsp to atmosphere folder and change title id in loader.ini.
I use Zelda Breath of the wild eur game title id : 01007EF00011E000
My loader.ini is
[config]
hbl_tid=01007EF00011E000
hbl_path=atmosphere/hbl.nsp
override_key=R

### Crash report?

No

### Environment?

- What bootloader (fusee, hekate, etc) was Atmosphère launched by: fusee-primary.bin
- Official release or unofficial build: official
- Do you have additional kips you're loading: I made a FS.kip1 in the kips to load the exfat driver.
I put exefs_patches and kip_patches from v 10.1 of SDFilesSwitch in the atmosphere folder
- Additional info about your environment:

### Additional context?

yellows8, 2018-10-28:
"black ribbon" What. That's just the temporary logo screen right?

What hbmenu build are you using?

yellows8, 2018-10-28:
This is not an atmo* issue. It's temporary, ignore it & wait for the next hbmenu release.