Atmosphere-NX/Atmosphere #250: Feature: 3DS controller integration

NitroCipher, 2018-10-29:
## Feature Request
Add ability to connect a 3DS system as a "Pro Controller", in conjunction with a homebrew application on the 3DS. (or even less likely, using the pre-existing "Wii U Smash Controller" app, as to not require the users 3DS to have CFW)

### Why would this feature be useful?
This would for users to expand the range of controllers supported by the system, and would be useful for party games such as Mario Kart or the upcoming Smash Bros title.

ThatNerdyPikachu, 2018-10-29:
i don't know if this is satire or not

NitroCipher, 2018-10-29:
@ThatNerdyPikachu Yeah, sorry if this idea is stupid. :/ But no, this wasn't meant as satire

SciresM, 2018-10-30:
This sounds like a really good user-mode homebrew -- it's a bit out of scope for ams, though.

closed 2018-10-30 by SciresM

NitroCipher, 2018-10-30:
@SciresM Alright. I assumed it would have needed to be through CFW. Didn't realize it could be done with user-mode stuff