Atmosphere-NX/Atmosphere #251: Is creport actually working?

marco-calautti, 2018-10-29:
## Bug Report

### What's the issue you encountered?
Errors seems to be still logged to flash memory, even though creport is correctly placed in its titleid directory. Going to System settings -> Health and Safety -> Error History, I can see a 2144-0001 error recorded.

### How can the issue be reproduced?
My sd just contains the atmosphere directory structure with fusee-seconday.bin+BCT.INI in the root, exefs.nsp in atmosphere/titles/0100000000000036 and hbl.nsp+loader.ini in /atmosphere/.
### Crash report?

(If a crash report was created under /atmosphere/crash_reports/, please upload it to 
[gist](https://gist.github.com/) and paste the link here.)

No, this indeed the issue, crash reports don't seem to be created on the sd, rather on the internal memory.
### Environment?

- What bootloader (fusee, hekate, etc) was Atmosphère launched by:
Fusee-primary.bin
- Official release or unofficial build:
official 0.7.3 build.
- Do you have additional kips you're loading:
sys-ftp.kip only
- Additional info about your environment:
FW 6.0.1

### Additional context?

yellows8, 2018-10-29:
creport is only for *crash* reports.

marco-calautti, 2018-10-29:
> creport is only for _crash_ reports.

Right, so this does not count as a crash. As far as I remember, 2144-0001 is an error code relating to libnx. So does it mean that the real only way we have to prevent Nintendo finding out I am using homebrew software is via disabling eclct? Hence, I will never be able to boot my Switch without Atmosphere running?

fincs, 2018-10-29:
It is not. libnx errors are 2345-xxxx

marco-calautti, 2018-10-29:
> It is not. libnx errors are 2345-xxxx.
> As long as you have disabled 'Send error reports' under system settings you should be fine, even on stock.

Perfect, thanks for the clarification!

The0919, 2018-10-30:
I also had a similar experience, but the error was 2345-0021 and it was a full system crash

fincs, 2018-10-30:
That's LibnxError_BadGfxDequeueBuffer. Are you running an application compiled with very old libnx (1.3.x or earlier)? If so, then the application needs to be recompiled with latest libnx.

SciresM, 2018-10-30:
Closing either way, since this isn't an ams issue.

closed 2018-10-30 by SciresM