Atmosphere-NX/Atmosphere #253: Title ID redirection not working when atmosphere booted using kips/hekate

fennectech, 2018-10-30:
i am currently loading atmosphere using hekate ctcaer mod  
[Atmosphere With FTP]
secmon=modules/atmosphere/exosphere.bin
kip1=modules/atmosphere/loader.kip
kip1=modules/atmosphere/pm.kip
kip1=modules/atmosphere/sm.kip
kip1=modules/atmosphere/fs_mitm.kip
kip1=modules/sysftpd/sys-ftpd.kip
kip1patch=nosigchk
logopath=bootlogo.bmp
atmosphere=1
This is how its being loaded   Everything works properly but the nsp titleid redirection wich results a hang at the nintendo switch loading screen and a crash after a minute or so.

SciresM, 2018-10-30:
...Does it work when booted through fusee-primary?

fennectech, 2018-10-30:
It doesn’t.   Same error

SciresM, 2018-10-30:
Could you give details on the redirection? Are you trying to replace something with HBL? What does your loader.ini look like? Have you updated your HBL recently? Older HBL lack support for application memory configuration.

SciresM, 2018-10-30:
Actually, why did you not use the issue templates?

fennectech, 2018-10-30:
It must have been selected when i started typing into the box       I have HBL updated to latest from the github as of yesterday   same with hbmenu    Im trying to replace a title (kitten squad at current wich ive gotten to work in the past but that was before atmosphere released   it hasent worked since 0.7.0 (wich was the first time i updated in a while)

SciresM, 2018-10-30:
"What does your loader.ini look like?"

fennectech, 2018-10-30:
whoops   Sorry   
\﻿[config]
;hbl_tid=010000000000100D
hbl_tid=01000C900A136000
hbl_path=atmosphere/hbl.nsp
override_key=!R

SciresM, 2018-10-30:
1 -- remove the comment line, comments aren't supported.

 2. Are you sure you have an official HBL release? Reminder that atmosphere HBL support is for exefs NSPs, not warez NSPs (which are NCAs inside of a PFS0).

If you're using an unofficial build you may be using a warez NSP instead of an exefs NSP, which would cause a crash.

fennectech, 2018-10-30:
Im using a 44 kilobyte NSP obtained from here
https://github.com/switchbrew/nx-hbloader/releases

fennectech, 2018-10-30:
i have no /atmosphere/kips    atmosphere/titles has mods for unrelated games (plus creport)

yellows8, 2018-10-30:
hbmenu v3.0.0?

fennectech, 2018-10-30:
Yep
Same happened with 2.0.0 tho

fennectech, 2018-10-30:
And   it happens with bedrock when booting with fusee-primary.bin

yellows8, 2018-10-30:
"a crash " details

fennectech, 2018-10-30:
it hangs on the spinning nintendo switch logo for about 45 seconds to a minute   then gives a generic nintendo switch crash screen   Ill get a screenshot

fennectech, 2018-10-30:
[image]

fennectech, 2018-10-30:
[01540885330_0100000000000023.log](https://github.com/Atmosphere-NX/Atmosphere/files/2528513/01540885330_0100000000000023.log)

SciresM, 2018-10-30:
Could you try with the official 0.7.3 build instead of unofficial ones?

fennectech, 2018-10-30:
where could i get kips for 0.7.3?  (ill try the official one with fusee-primary now tho)

SciresM, 2018-10-30:
This sounds like it's *probably* a HBL issue and not an atmosphere issue. Hard to tell, though... @fincs @yellows8?

SciresM, 2018-10-30:
@fennectech could you paste the output of tree /F /A run inside sdmc:/atmosphere/?

fennectech, 2018-10-30:
https://pastebin.com/yi54ejM9

yellows8, 2018-10-30:
"a HBL issue" doubt it

yellows8, 2018-10-30:
Try with a clean atmo* dir where the only change is hbl.nsp/loader.ini.

fennectech, 2018-10-30:
Nope   Same issue.

yellows8, 2018-10-31:
Can you reformat SD / use a different card?

fennectech, 2018-10-31:
i dont have any that are free tor format :/

fennectech, 2018-10-31:
scratch that   found an 8 gig

SciresM, 2018-10-31:
Got reports that this reproduces for BotW 1.5.0 on sysver 5.1.0. Interestingly, does *not* repro for BotW 1.0.0.

SciresM, 2018-10-31:
@yellows8 I suspect there's something going wrong NPDM-wise. Mmheap/pool partition related, maybe?

SciresM, 2018-10-31:
This isn't an atmosphere bug, it's caused by "Automatic" mode video capture recording causing AM to try to allocate 96 MB of memory that HBL has already claimed.

fennectech, 2018-10-31:
I will test this by assigning it hulu wich has that disabled

SciresM, 2018-10-31:
@fennectech this isn't an atmosphere bug, it's a HBL issue. It'll get fixed.

closed 2018-10-31 by SciresM

fennectech, 2018-10-31:
well   That was a fun rabbit hole!     At least now i wont be accidently closing RA with a bump of the home screen