Atmosphere-NX/Atmosphere #256: commit 9142e90a048faee4dcc0cee28e0ddc531df416b3 crash fs_mitm set_mitm

Ninoh-FOX, 2018-11-06:
## Bug Report

### What's the issue you encountered?

games with layered crash the first boot, the second boot run the "normal" game

### How can the issue be reproduced?

### Crash report?

(If a crash report was created under /atmosphere/crash_reports/, please upload it to 
[gist](https://gist.github.com/) and paste the link here.)

```
Atmosphère Crash Report (v1.1):
Result:                          0x25A0B (2011-0301)

Process Info:
    Process Name:                Application
    Title ID:                    0100d870045b6000
    Process ID:                  000000000000007f
    Process Flags:               00000077
    User Exception Address:      0000003360fd2000
Exception Info:
    Type:                        User Break
    Address:                     0000004136f67810 (nnSdk + 0xd9810)
    Break Reason:                0x0
    Break Address:               000000193609ee28
    Break Size:                  0x4
Crashed Thread Info:
    Thread ID:                   00000000000001ed
    Stack:                       0000001935f9f000-000000193609f000
    Registers:
        X[00]:                   0000000000000000
        X[01]:                   0000000000000000
        X[02]:                   0000000000000000
        X[03]:                   0000000000000000
        X[04]:                   0000000000000000
        X[05]:                   0000000000000000
        X[06]:                   0000000000000000
        X[07]:                   0000000000000000
        X[08]:                   0000000000000000
        X[09]:                   0000000000000000
        X[10]:                   0000000000000000
        X[11]:                   0000000000000000
        X[12]:                   0000000000000000
        X[13]:                   0000000000000000
        X[14]:                   0000000000000000
        X[15]:                   0000000000000000
        X[16]:                   0000000000000000
        X[17]:                   0000000000000000
        X[18]:                   0000000000000000
        X[19]:                   0000000000000000
        X[20]:                   0000000000000000
        X[21]:                   0000000000000000
        X[22]:                   0000000000000000
        X[23]:                   0000000000000000
        X[24]:                   0000000000000000
        X[25]:                   0000000000000000
        X[26]:                   0000000000000000
        X[27]:                   0000000000000000
        X[28]:                   0000000000000000
        FP:                      000000193609ec80
        LR:                      0000004136edbd0c (nnSdk + 0x4dd0c)
        SP:                      000000193609ec80
        PC:                      0000004136f6780c (nnSdk + 0xd980c)
    Stack Trace:
        ReturnAddress[00]:       0000004136eda780 (nnSdk + 0x4c780)
        ReturnAddress[01]:       0000004136eda844 (nnSdk + 0x4c844)
        ReturnAddress[02]:       0000004136edd59c (nnSdk + 0x4f59c)
        ReturnAddress[03]:       0000004136f054dc (nnSdk + 0x774dc)
        ReturnAddress[04]:       0000004133e23ea0 (l-classics-app.nss + 0x1fea0)
        ReturnAddress[05]:       0000004133e04280 (l-classics-app.nss + 0x280)
        ReturnAddress[06]:       0000004133e04620 (l-classics-app.nss + 0x620)
        ReturnAddress[07]:       0000004136f1292c (nnSdk + 0x8492c)
        ReturnAddress[08]:       0000004133e000ac (nnrtld + 0xac)
Code Region Info:
    Number of Code Regions:      5
    Code Region 00:
        Address:                 0000004133e00000-0000004133e02000
        Name:                    nnrtld
        Build Id:                E701850BDB94823AFB3C0F133137832BF0E8D666000000000000000000000000
    Code Region 01:
        Address:                 0000004133e04000-000000413416c000
        Name:                    l-classics-app.nss
        Build Id:                877ABB83FCF6B41E44BFB42F03684D6500000000000000000000000000000000
    Code Region 02:
        Address:                 000000413596c000-0000004136142000
        Name:                    glslc
        Build Id:                2FA6E9856F72565F0B1C0BDBCC2B2C221D356D45000000000000000000000000
    Code Region 03:
        Address:                 0000004136796000-0000004136ad8000
        Name:                    multimedia
        Build Id:                2E8D3A967B3539EB77F50D1647380378CD481ACF000000000000000000000000
    Code Region 04:
        Address:                 0000004136e8e000-0000004137376000
        Name:                    nnSdk
        Build Id:                F7C7B7A90ECDC2D41C97508CC7236B8AF973B76C000000000000000000000000

Thread Report:
Number of Threads:               01
Threads[00]:
    Thread ID:                   00000000000001ed
    Stack:                       0000001935f9f000-000000193609f000
    Registers:
        X[00]:                   0000000000000000
        X[01]:                   0000000000000000
        X[02]:                   0000000000000000
        X[03]:                   0000000000000000
        X[04]:                   0000000000000000
        X[05]:                   0000000000000000
        X[06]:                   0000000000000000
        X[07]:                   0000000000000000
        X[08]:                   0000000000000000
        X[09]:                   0000000000000000
        X[10]:                   0000000000000000
        X[11]:                   0000000000000000
        X[12]:                   0000000000000000
        X[13]:                   0000000000000000
        X[14]:                   0000000000000000
        X[15]:                   0000000000000000
        X[16]:                   0000000000000000
        X[17]:                   0000000000000000
        X[18]:                   0000000000000000
        X[19]:                   0000000000000000
        X[20]:                   0000000000000000
        X[21]:                   0000000000000000
        X[22]:                   0000000000000000
        X[23]:                   0000000000000000
        X[24]:                   0000000000000000
        X[25]:                   0000000000000000
        X[26]:                   0000000000000000
        X[27]:                   0000000000000000
        X[28]:                   0000000000000000
        FP:                      000000193609ec80
        LR:                      0000004136edbd0c (nnSdk + 0x4dd0c)
        SP:                      000000193609ec80
        PC:                      0000004136f6780c (nnSdk + 0xd980c)
    Stack Trace:
        ReturnAddress[00]:       0000004136eda780 (nnSdk + 0x4c780)
        ReturnAddress[01]:       0000004136eda844 (nnSdk + 0x4c844)
        ReturnAddress[02]:       0000004136edd59c (nnSdk + 0x4f59c)
        ReturnAddress[03]:       0000004136f054dc (nnSdk + 0x774dc)
        ReturnAddress[04]:       0000004133e23ea0 (l-classics-app.nss + 0x1fea0)
        ReturnAddress[05]:       0000004133e04280 (l-classics-app.nss + 0x280)
        ReturnAddress[06]:       0000004133e04620 (l-classics-app.nss + 0x620)
        ReturnAddress[07]:       0000004136f1292c (nnSdk + 0x8492c)
        ReturnAddress[08]:       0000004133e000ac (nnrtld + 0xac)
```

### Environment?

- What bootloader (fusee, hekate, etc) was Atmosphère launched by: fusee and hekate
- Official release or unofficial build: unofficial build
- Do you have additional kips you're loading: no
- Additional info about your environment: hekate config: 
[Atmosphere hetake]
secmon=modules/cfw/thermosphere.bin
secmon=modules/cfw/exosphere.bin
kip1=modules/cfw/loader.kip
kip1=modules/cfw/sm.kip
kip1=modules/cfw/pm.kip
kip1=modules/cfw/fs_mitm.kip
kip1=modules/nx-dreport.kip
fullsvcperm=1
atmosphere=1
kip1patch=nosigchk

fusee config I load simply fusee-primary.bin

### Additional context?

if you compile the code with the old library for libstratosphere, everything works correctly

Ninoh-FOX, 2018-11-07:
Well... Maybe hetake is bad. But if I load fusee-primary.bin I have the same error, the game is the Nintendo online emulator (oficial) + update.

SciresM, 2018-11-07:
@Ninoh-FOX Do you know what specific commit broke things? Are you available on ReSwitched's discord? I'm interested in debugging this...

SciresM, 2018-11-07:
@Ninoh-FOX If fs.mitm was active, it should have created a `romfs_metadata.bin` under atmosphere/titles/<nes classic tid>/` -- could you upload that? Newer commits make fs.mitm not even attempt to replace romfs if there's nothing to mitm, so the folder should exist...

Ninoh-FOX, 2018-11-07:
the new file that I have is METADATALOG.bin in the root sd. I have discord but I don´t have that channel.

SciresM, 2018-11-07:
@Ninoh-FOX So the change is specifically the one that switched WaitableManager to use Intra-process events for new waitable signaling?

SciresM, 2018-11-07:
Also, it looks like you're modding super heavily, which is relevant...I really am interested in helping debug but you'll have to contact me on discord.

SciresM, 2018-11-07:
@Ninoh-FOX Can you try again with the latest commit? I think it should fix a race condition related to adding new waitables.

Ninoh-FOX, 2018-11-07:
I go to a compiler that now

SciresM, 2018-11-07:
Also, please do join [the ReSwitched discord](https://discordapp.com/invite/DThbZ7z), and PM me for further debugging if needed.

Ninoh-FOX, 2018-11-07:
crash...

`Atmosphère Crash Report (v1.1):
Result:                          0x25A0B (2011-0301)

Process Info:
    Process Name:                Application
    Title ID:                    0100d870045b6000
    Process ID:                  0000000000000080
    Process Flags:               00000077
    User Exception Address:      00000057c776b000
Exception Info:
    Type:                        User Break
    Address:                     00000032d0567810 (nnSdk + 0xd9810)
    Break Reason:                0x0
    Break Address:               0000006d30550e28
    Break Size:                  0x4
Crashed Thread Info:
    Thread ID:                   00000000000001ed
    Stack:                       0000006d30451000-0000006d30551000
    Registers:
        X[00]:                   0000000000000000
        X[01]:                   0000000000000000
        X[02]:                   0000000000000000
        X[03]:                   0000000000000000
        X[04]:                   0000000000000000
        X[05]:                   0000000000000000
        X[06]:                   0000000000000000
        X[07]:                   0000000000000000
        X[08]:                   0000000000000000
        X[09]:                   0000000000000000
        X[10]:                   0000000000000000
        X[11]:                   0000000000000000
        X[12]:                   0000000000000000
        X[13]:                   0000000000000000
        X[14]:                   0000000000000000
        X[15]:                   0000000000000000
        X[16]:                   0000000000000000
        X[17]:                   0000000000000000
        X[18]:                   0000000000000000
        X[19]:                   0000000000000000
        X[20]:                   0000000000000000
        X[21]:                   0000000000000000
        X[22]:                   0000000000000000
        X[23]:                   0000000000000000
        X[24]:                   0000000000000000
        X[25]:                   0000000000000000
        X[26]:                   0000000000000000
        X[27]:                   0000000000000000
        X[28]:                   0000000000000000
        FP:                      0000006d30550c80
        LR:                      00000032d04dbd0c (nnSdk + 0x4dd0c)
        SP:                      0000006d30550c80
        PC:                      00000032d056780c (nnSdk + 0xd980c)
    Stack Trace:
        ReturnAddress[00]:       00000032d04da780 (nnSdk + 0x4c780)
        ReturnAddress[01]:       00000032d04da844 (nnSdk + 0x4c844)
        ReturnAddress[02]:       00000032d04dd59c (nnSdk + 0x4f59c)
        ReturnAddress[03]:       00000032d05054dc (nnSdk + 0x774dc)
        ReturnAddress[04]:       00000032cd423ea0 (l-classics-app.nss + 0x1fea0)
        ReturnAddress[05]:       00000032cd404280 (l-classics-app.nss + 0x280)
        ReturnAddress[06]:       00000032cd404620 (l-classics-app.nss + 0x620)
        ReturnAddress[07]:       00000032d051292c (nnSdk + 0x8492c)
        ReturnAddress[08]:       00000032cd4000ac (nnrtld + 0xac)
Code Region Info:
    Number of Code Regions:      5
    Code Region 00:
        Address:                 00000032cd400000-00000032cd402000
        Name:                    nnrtld
        Build Id:                E701850BDB94823AFB3C0F133137832BF0E8D666000000000000000000000000
[... 177 lines cut ...]
        X[19]:                   0000000000000000
        X[20]:                   0000000000000000
        X[21]:                   0000000000000000
        X[22]:                   0000000000000000
        X[23]:                   0000000000000000
        X[24]:                   0000000000000000
        X[25]:                   0000000000000000
        X[26]:                   0000000000000000
        X[27]:                   0000000000000000
        X[28]:                   0000000000000000
        FP:                      0000006d30550c80
        LR:                      00000032d04dbd0c (nnSdk + 0x4dd0c)
        SP:                      0000006d30550c80
        PC:                      00000032d056780c (nnSdk + 0xd980c)
    Stack Trace:
        ReturnAddress[00]:       00000032d04da780 (nnSdk + 0x4c780)
        ReturnAddress[01]:       00000032d04da844 (nnSdk + 0x4c844)
        ReturnAddress[02]:       00000032d04dd59c (nnSdk + 0x4f59c)
        ReturnAddress[03]:       00000032d05054dc (nnSdk + 0x774dc)
        ReturnAddress[04]:       00000032cd423ea0 (l-classics-app.nss + 0x1fea0)
        ReturnAddress[05]:       00000032cd404280 (l-classics-app.nss + 0x280)
        ReturnAddress[06]:       00000032cd404620 (l-classics-app.nss + 0x620)
        ReturnAddress[07]:       00000032d051292c (nnSdk + 0x8492c)
        ReturnAddress[08]:       00000032cd4000ac (nnrtld + 0xac)
`

SciresM, 2018-11-08:
Found issue + fixed, works when I test locally. Please compile + confirm? :)

SciresM, 2018-11-08:
@Ninoh-FOX ^

Ninoh-FOX, 2018-11-08:
congratulations!! now works fine!!! ^W^

commit bac81f4c (2018-11-08): libstrat: fix potential oob deref in WaitableManager (closes #256)