Atmosphere-NX/Atmosphere #259: build failure

fennectech, 2018-11-09:
### What's the issue you encountered?
Attempt to build atmosphere fails with the following.

make[1]: Leaving directory '/home/fennectech/Atmosphere/exosphere'
make -C stratosphere all
make[1]: Entering directory '/home/fennectech/Atmosphere/stratosphere'
make -C libstratosphere all
make[2]: Entering directory '/home/fennectech/Atmosphere/stratosphere/libstratosphere'
make[2]: *** No rule to make target 'all'.  Stop.
make[2]: Leaving directory '/home/fennectech/Atmosphere/stratosphere/libstratosphere'
make[1]: *** [Makefile:12: libstratosphere] Error 2
make[1]: Leaving directory '/home/fennectech/Atmosphere/stratosphere'
make: *** [Makefile:18: stratosphere] Error 2

### How can the issue be reproduced?

### Environment?
Linux
All software from dkp-pacman is up to date as of nov-9 4:00AM

SciresM, 2018-11-09:
update your submodules.

closed 2018-11-09 by SciresM

fennectech, 2018-11-09:
how do i do this?     Its a freashly cloned repo

SciresM, 2018-11-09:
`git submodule update --init --recursive`

just do your clones with -r in the future.

You'll likely need to learn how to use submodules if you want to continue building ams in the future.

fennectech, 2018-11-09:
okay. Thank you.

fennectech, 2018-11-09:
Still fails with this now.   https://gist.github.com/fennectech/1607a48bc542a521799d7735af24a4ab
Would this be something thats opened in a seperate issue or continued in this one?

SciresM, 2018-11-09:
This isn't an atmosphere issue. This is a "your libnx issue is out of date" issue..........you shouldn't be commenting about this at all here, use ReSwitched discord #user-support or #switchdev on EFnet. :/

fennectech, 2018-11-09:
thanks   Ill compile libnx and install from the git

SciresM, 2018-11-09:
Your gist mentions DomainResponseHeader not naming a type...this just indicates your build environment is bad, as it's [clearly defined](https://github.com/switchbrew/libnx/blob/master/nx/include/switch/kernel/ipc.h#L70) in libnx (and has been for a while)...

fennectech, 2018-11-09:
after doing make && make install in the libnx repo all seems to be working now  (its not bailing within a few seconds any more)   Thanks

fincs, 2018-11-09:
Please do not build and install libnx from git unless we specifically tell you to. There should be nothing preventing you from using the latest stable version of libnx to build Atmosphère. It's likely you had an outdated version; upgrading is as easy as running `pacman -Syu`. Unfortunately you have now corrupted your toolchain install by changing the permissions on it and installing a development snapshot, and will need to start from scratch.