Atmosphere-NX/Atmosphere #260: Disable HDCP for media streaming apps

fennectech, 2018-11-09:
## Feature Request

### What feature are you suggesting?
I would like to see an ability to disable HDCP on the switch.  Its not always on so it should be possable to make it not enable at all.  
The reason is some cheaper TVs work fine for games but dont play nice with the switch's implimentation of HDCP

sigboe, 2018-11-09:
I am just speculating here.

It's possible that, the HDCP is managed by a chip that is in between the GPU and video output, and may not be feasible to bypass this in custom firmware.

fennectech, 2018-11-09:
Most apps dont have HDCP enabled and i only encounter issues with it when opening a streaming app (hulu or youtube)   so there has to be a way to turn it off from software.  Becuase its not always on.

sigboe, 2018-11-09:
Okay, I didn't know that, I haven't updated my switch since 4.1

Then I believe it to be possible at least.