Atmosphere-NX/Atmosphere #262: Atmosphere version

jul2003, 2018-11-10:
## Feature Request

### What feature are you suggesting?
Display atmosphere version in system menu.
This works on 0.7.3 but not on 0.7.4

### Why would this feature be useful?
Know if the console is started on atmosphere or OFW

fincs, 2018-11-10:
This works on 0.7.4. Did you extract the entire zip archive to the root of the SD card?

jul2003, 2018-11-10:
yes i did that my ofw is 4.1.0

jul2003, 2018-11-10:
Ok small update.
My sdcard is exfat and my switch doesn't have the exfat driver so i made a kip file to add exfat driver on atmosphere.
I tried to format the sdcard on fat filesystem so no need to load the driver and now i have the ams version in system menu
I put back the exfat driver in the kips folder of atmosphere and i still have ams version.
From the switch i format the sd card then it's formatted into exfat.
I put again all the files for atmosphere on the sd and the exfat driver and then no ams version in the system menu.
So it's related with the fact the sd is in exfat i assume

closed 2018-11-10 by SciresM

jul2003, 2018-11-11:
issue closed ok but can you tell what i am doing wrong and what i have to do ? because in 0.7.3 in the same configuration my sd in exfat and the exfat driver loaded by atmosphere the ams version is displayed in system menu. 
and in 0.7.4 if i don't put the driver on my exfat sd then ams doesn't load. and black screen at startup

SciresM, 2018-11-11:
Ams relies on the ability to read the SD card in userland. This means if you have an exfat card, you need exfat FIRM.

jul2003, 2018-11-17:
Hi,

I come back to you with this.

I updated the BCPKG2-1-Normal-Main.bin file of my switch to add the exFat support.
So now when i boot ofw my sdxc card is detected and i can format it from the switch into exfat.

Now if i launch atmosphere version 0.7.5 on the switch then the ams version is not displayed on the system menu.
Does that mean it's still not exfat ofw ?

If still not do you think there is a way to put my 4.1.0 firmware an exfat 4.1.0 like update the switch to 4.1.0 with ChoixDuJourNX or something ?

Thanks

jul2003, 2018-11-17:
Update : i revered the BCPKG2-1-Normal-Main.bin on my switch so i lost the exfat support.
With ChoixDuJourNX i updated the 4.1.0 firmware on my switch and i choose the exfat version of the 4.1.0. Now exfat is back on my switch and i can read my sd card.
When i run atmosphere 0.7.5 still the same the version is not displayed on the system.

If someone else with exfat sd card can confirm it works on not . Please. Thanks