Atmosphere-NX/Atmosphere #265: Bitmap Screenshots

NO-ob, 2018-11-13:
## Feature Request
BMP or  PNG Screenshots
### Why would this feature be useful?
The screenshots on the switch look awful as they are very low quality jpegs 
[image]
This is cropped but not by a whole lot just look at the artifacting

misson20000, 2018-11-13:
If someone figures it out and writes some docs on how to find relevant functions to patch I'll port it to other firmwares

boringhexi, 2018-12-29:
Seconding the request for lossless screenshots. (I’m making this request because The Cutting Room Floor, a site for documenting unused video game content, does not allow lossy JPEG screenshots, not even high-quality ones.) Even some sort of raw format that needs to be converted to BMP or PNG later would be fine by me.

SciresM, 2019-01-26:
Out of scope, for now.

closed 2019-01-26 by SciresM