Atmosphere-NX/Atmosphere #267: [Request] Alter shutdown temprature

fennectech, 2018-11-17:
I would love to be able to alter the temprature at wich the system goes to sleep   so that i can prevent it from getting too hot.  

if we decrease the temprature at wich the switch goes into overheat mode we could prevent the system from getting hot and shutdown at a more comfterble temprature

ZachyCatGames, 2018-11-18:
The Switch doesn't even get very hot though 🤔 (under normal circumstances)

fennectech, 2018-11-18:
The point of this is to limit the temperature at which it can get to in abnormal circumstances    Reducing the thermal override temp can increase the lifespan of the APU

Luro02, 2018-11-18:
There should also be an option to make the fan more sensitive (spins already on lower temperatures?!).

fennectech, 2018-11-18:
Sounds good to me!

ZachyCatGames, 2018-11-19:
@Luro02 I’ve played Xenoblade 2, it gets kinda warm/hot, but i wouldnt say it’s dangerously hot :P

fennectech, 2018-11-19:
my switch went into thermal shutdown while i was playing BOTW   id like to limit how hot it can get

yellows8, 2018-11-19:
"thermal shutdown while i was playing BOTW" That should *never* happen, has it happened on stock?

ZachyCatGames, 2018-11-19:
Well either:
A. Your switch doesnt have enough room
B. Your switch is a hot area
C. Your switch has a defect
or
D. You messed around with clockspeeds (which I doubt this is why)

fennectech, 2018-11-19:
It was not in a hot area.  As you can see there is plenty of room and it only happened once and hasent happened before or after that.  so i dont think its a defect.

fennectech, 2018-11-19:
And i dont think we even have the ability to alter the clock

SapphireExile, 2018-11-19:
If it was a one time thing, that could not be reproduced, why would someone waste development time implementing this?

fennectech, 2018-11-19:
Because the ability to alter the shutdown temprature is still a nice thing.

SapphireExile, 2018-11-19:
If it was a large problem, Nintendo would have set the limit appropriately. 
Artificially lowering the max temp (On a console that already has poor cooling) will just result in many more people having overheat issues. 
My room where my switch is regularly changes ambient temperature between 65 and 85 (Arduino temp + humidity sensor). If I were to lower the max temp, it would go off just because my room got hotter. 

Implementing something that modifies emergency behavior such as thermal shutdown isn't wise, and if used improperly, can result in a console that boots, then instantly turns off

fennectech, 2018-11-19:
Thats why its configured by a config file on the sdcard.    If you fuck it up you can delete the config file with an sdcard reader resetting it to default.   There isnt much harm you can do other than increasing it.   Also its something the user has to change via config file so only those who know what they are doing mess with it.    The

SapphireExile, 2018-11-19:
So let's say that it can be configured, and will actually be applied to hardware, what would be the point?
Let's say it has a maximum SoC temp of 100c. 
Now let's lower it to 90c. 

Getting from ambient (~22-27c) to high heat (70-80c) is usually easy. 

Getting from 80-90 is quite a bit harder. 
Getting to tmax (100) is extremely hard, and would require unique conditions, and probably fan failure. 
Allowing the console to reach 100c would be suicide, hence thermal shutdown. 
But getting to 92 can happen just because of a spike of dead air hanging around the exhaust, or my fatass cat blocking the exhaust. 
That second it takes to remove her would send me into thermal shutdown if it was modified to 90c, meaning losing all progress and potentially corrupting saves, data, among other things. 

Even if you reduce it to just 95 (still using the hypothetical 100c max), spikes in temp can cause shutdown. 

Reducing it to something absurd like 20% less (80c) would cause non-stop thermal shutdown, as most operating temps of electronics regularly hit this in normal usage. 

If ANYTHING, a throttle should be implemented. Fucking with emergency shutdown will never work, but throttling the SoC to allow it to cool without turning off the console would by far be the best option.

fennectech, 2018-11-19:
It doesn’t shut down.    It goes to sleep.  No save corruption or loss of progress.   Just wake the console up in a moment when its cooled off.

fennectech, 2018-11-19:
I agree.  Throttling would be even better

SapphireExile, 2018-11-19:
It'd be rather easy to implement as well. Simple force it into handheld mode clocks, wait for it to reach a threshold (Say 70c), then return to dock clocks.

fennectech, 2018-11-19:
That wont stop it from overheating if somehow the switch winds up running in a case.     Triggering the thermal protection at a reduced (user configurarable) temprature would allow preventing the SOC from reaching such high temperatures.  (It currently goes off at 100c i beleve.   Pulling it to 95 would do a lot for the lifespan of the SOC)

fennectech, 2018-11-19:
Bolth would be great tho

SapphireExile, 2018-11-19:
If the switch is at 100% usage in a case, something has gone very wrong. It should be sleeping, or at the very least, idling. 
Unless you mean a skin case, at which it should only be in handheld mode, and it automatically throttled by Nintendo themselves.

fennectech, 2018-11-19:
I’m talking about if the switch winds up in a zip case while the game is running.     For example a younger sibling puts your switch into the case without turning it off.     it could get verry hot verry fast.

hexkyz, 2018-11-19:
Something like this is extremely delicate to implement. Adding the ability to lower temperature thresholds or clock speeds would also make it much easier to do just the opposite.
Additionally, an official service (https://switchbrew.org/wiki/AM_services#tcap) to monitor these kind of events from userland was added in 5.0.0. It may be a better idea to try exploring and interacting with it instead.

closed 2018-11-19 by SciresM