Atmosphere-NX/Atmosphere #269: Moonlighter error at launch

Zriel88, 2018-11-18:
## Bug Report
### What's the issue you encountered?
Moonlighter game does not boot with atmosphere 0.7.5
### How can the issue be reproduced?
Install NSP from moonlighter and moonlighter update (v131072).
Launch game.
Switch will report there is corrupt game data
can share NSPs in a gdrive link if needed.
### Crash report?
No crash

### Environment?
- What bootloader (fusee, hekate, etc) was Atmosphère launched by: hetake-ctaer 4.2
- Official release or unofficial build: 0.7.5 from SDFiles/Kosmos
- Do you have additional kips you're loading: None
- Additional info about your environment: 

### Additional context?
Uploaded the issue here because the game is working correctly under ReinX. I don't really know if I'm breaking any rule here, just wanted to have the devs known of this issue (in case they are not aware of)

Skonikol, 2018-11-19:
I can confirm that some users of version 075, running through hekate42, there is a problem with this game. After launching on consoles users AMS via fuse 1-2 game included correctly.

misson20000, 2018-11-19:
Does it work if you buy the game

hexkyz, 2018-11-19:
I've edited your original report solely to remove the allusion to sharing of pirated content. In the future, please refrain from adding such remarks.

Could you please take a picture of each error message displayed on the Switch?

Zriel88, 2018-11-19:
Thanks for the heads up.
First try to run
https://imgur.com/EMVdHX8
Translates to 
"Couldn't launch the program
Try it again from the HOME menu"

And second try to run after that.
https://imgur.com/BTf8V7m

Translates to
"Damaged data has been found
It is recommended to use the function "Search for damaged data"
Cancel/ Search"

tiliarou, 2018-11-20:
The switch OS found corrupted data. This looks like it's not atmosphere issue... You should ask on discord or open an issue on whatever CFW pack you use (anyway you shouldn't be able to install the game with stock atmo).

Zriel88, 2018-11-20:
Ok, thanks for the info.

closed 2018-11-27 by SciresM