Atmosphere-NX/Atmosphere #271: Custom splash

Skonikol, 2018-11-19:
## Bug Report

I'm trying to launch a custom splash with the settings:

_BCT0
[stage1]
stage2_path = fusee-secondary.bin
stage2_addr = 0xF0000000
stage2_entrypoint = 0xF0000000
[stage2]
custom_splash = splash.bmp_

The logo works correctly when starting hekate.
splash.bmp is in the sd root 
### What's the issue you encountered?
The standard logo is launched.
### How can the issue be reproduced?
Run my logo with my BCT settings (but I don't know where to download BMP).
### Crash report?
no
### Environment?

- What bootloader (fusee, hekate, etc) was Atmosphère launched by: **fusee**
- Official release or unofficial build: **Official release**
- Do you have additional kips you're loading: **fs_patches by The-4n, es_patches by Rajkosto**
- Additional info about your environment: **OWF 6.1.0 (one fuse is lost)**
### Additional context?
no

hexkyz, 2018-11-19:
There was a bug in reading the splash screen bitmap from the SD card. Could you please try again with the latest changes?

Skonikol, 2018-11-19:
> There was a bug in reading the splash screen bitmap from the SD card. Could you please try again with the latest changes?
Yeah, everything works fine now. Thank you for solving the problem so quickly.

Skonikol, 2018-11-19:
Works

Skonikol, 2018-11-19:
Small question. Is there any way to adjust the delay?

hexkyz, 2018-11-19:
> 
> 
> Yeah, everything works fine now. Thank you for solving the problem so quickly.

Awesome! Thanks for testing!

> 
> 
> Small question. Is there any way to adjust the delay?

No, not yet, but this can be easily implemented. I need to run some tests first just to make sure changing the delay doesn't introduce any timing issues.

closed 2018-11-20 by Skonikol