Atmosphere-NX/Atmosphere #273: possible issue with layeredFS and changing the hot key for HB menu...

gravysamich, 2018-11-20:
i am playing lets go eevee. i wanted to use 2 mods. 1 changes eevee's skin to the shiny version the other makes all 151 pokemon available to catch.

the mods load fine so long as everything with atmosphere is default. however i like to gave the gallery open by default but access the homebrew menu when i hold ZR. i made the change with the loader.ini to get this effect. as soon as i made this change my mods no longer work. they just refuse to load. if i revert to the default, everything is fine again.

i am using the atmosphere loader that i downloaded here, the fusee-primary.bin.

Skonikol, 2018-11-20:
Hold ZR when you start the game and the mods will be enabled.

futcharist, 2018-11-20:
I am having the same issue. I had layeredFS mods loading this morning, before making a change to atmosphere's homebrew shortcut. I have tried every permutation of fixing my own problem, including formatting my SD card in both formats, letting the switch format my SD card and creating all new files. I even loaded up my NAND backup, then re-modified, essentially re-hacking my switch from scratch.  I even attempted to revert my save to when I knew mods were working, but no go. At this point, LayeredFS is not functioning AT ALL and I am at a loss for why it should not be. My installation is as fresh as it gets, right off of sdsetup.com

gravysamich, 2018-11-21:
worked like a charm! 

is this a bug or a feature then? 

i could see a few instances when this would be useful.

SciresM, 2018-11-21:
@gravysamich this is a feature -- the override key in loader.ini controls *all content overriding*, both for loader and for fs.mitm.

closed 2018-11-21 by SciresM