Atmosphere-NX/Atmosphere #274: switch between layeredFS folders with holding a button when starting a game

ShinKaz, 2018-11-27:
Ok, so I guess this sounds a bit similar as the Multi-LayeredFS or per user LayeredFS but hear me out ;)

just like the way HBL works, you'd be able to choose the LayeredFS folder/patch you want by holding R(or another button) right after you start your game/app

(my case use is my copy of taiko n t, you can patch soungs from the wiiu games, and soon ps4 and probably more, but there's a 200 max song limit)

so holding R would start the game with titles/0100D9A005ED6000[R]/ layeredFS folder
same with [L] [Y] [X] [B] [P](lus) [M](inus) [U]p [D]own

you could even have a bunch of titles/titleidxyz[btn]/ folders but no titles/titleidxyz/ folder for that game, enable you to just start the game clean by hitting A. (great if your system is used by kids for example) (no need to restart the switch in none layeredFS mode (tricky for a child if autorcm is on) or mess with renaming a file folder)

if titles/titleidxyz/ folder was still there, it would start normally like it does now, not breaking the compatibility.

This feature would be useful because the switch is a household device, and ppl would want to switch between the different patches by simply holding a button, without using homebrew like game mods managers or renaming folder with nx-shell etc, it would also allow to be able to start the clean version of the game without restarting the game in "normal"/no "layeredFS mode".

SapphireExile, 2018-11-27:
I'm pretty sure there's already Homebrew that does this

SapphireExile, 2018-11-28:
Regular end users aren't using layeredFS in the first place. 
Those that are know how to use it, and how to manage homebrew. 

What you want to do would require game specific programming (read: Homebrew), game specific folders (../game/layer1 layer2 layer3 etc), and game specific hotkeys for switching (Would Y load layer1 or 3?).

How would it know what to add and what to omit? Folders would have to be specifically named (Game_Layer1_xxx), and the content inside would have to be made specifically for it because what you want to do is more like DLC injection instead of filesystem management. 

You also want it to be able to run normally, which would require forcing layeredFS to not activate. 

To summarize, what you want done would require Atmosphere to support the specific game, and give the game's specific homebrew access to controlling layeredFS. 

While we have homebrew that allows you to switch which layer you want to use, binding it to a simple button seems to be out of scope. LayeredFS as a whole would have to be modified to do this.

ShinKaz, 2018-11-28:
the person setting up the files and the one using don't have to be the same person, specifically.

I don't understand what you are talking about, the patches I'm using right now (so yeah, they're working, nothing needs to change for them to work) they don't need DLCs injection.

you're asking how to match layer1 with Y and layer2 with R when I never even suggested such a thing.
with not call folders :
titles/0100D9A005ED6000[Y]/ (put files for your layer"1" there)
titles/0100D9A005ED6000[R]/ (put files for your layer"2" there)

if titles/0100D9A005ED6000/ is no there, you can then start the clean game by hitting A. (no need to restart the cfw(which a normal end user or child probably can't do anyways))

layerFS would not active if there was no folder for that game in titles, like it does now.

what do you mean how would it know how to add or omit ? err...I don't know, like it does now ?
just load one folder not another one ? gee, how layeredFS know how to load titles/0100D9A005ED6000/ instead of titles/01007EF00011E000/ ? I wonder :o if it can read chars, it can read [R] too. and holding or any button while opening a chosen title for HBL works too.
I made a request because most of those function are already in Atmosphère. 

like I described I don't see how code has to be integrated for that game specifically to do this, does it now ? yet it works.

SciresM, 2018-11-28:
While @SapphireExile's description is wrong, this doesn't jive with supporting non-override content. Right now, both layeredfs and loader share a common override key, which is user-controllable. I can't really see how to reconcile "user-controlled override toggle key, including support for on-by-default" with "button combinations used to decide what folder to read contents from".

Gonna close this due to infeasibilty, but feel free to continue discussing -- if there's some way to reconcile, I'm not opposed.

closed 2018-11-28 by SciresM