Atmosphere-NX/Atmosphere #275: Games that display login screens after starting don't start at all

bananapi761, 2018-11-27:
## Bug Report

### What's the issue you encountered?
Games that display login screens after starting don't start at all
### How can the issue be reproduced?
Loading any installed game that displays a login screen after it starts (shows 'Could not start the software' instantly)
### Crash report?

(If a crash report was created under /atmosphere/crash_reports/, please upload it to 
[gist](https://gist.github.com/) and paste the link here.)

### Environment?

- What bootloader (fusee, hekate, etc) was Atmosphère launched by: fusee
- Official release or unofficial build: official 0.7.5 with sigpatches
- Do you have additional kips you're loading: No
- Additional info about your environment: 6.1 FW, updated officially

### Additional context?

SciresM, 2018-11-27:
@bananapi761 there is insufficient information for me to act on this, here. Is there a crash report? Could you mention the specific title(s) this occurs for?

fennectech, 2018-11-27:
Are you talking about the user picker?

Skonikol, 2018-11-28:
I'm assuming damage to the Nintendo folder. Perhaps you removed the archive attributes from it entirely. If the games installed in nand behave the same way, then my guess is wrong.

fennectech, 2018-11-28:
Can you use the check game data thing in the system settings?

bananapi761, 2018-11-28:
The console itself doesn't crash, just the game with a 'Could not start the software' message. It happens with games like 1-2 Switch and Yooka-laylee. They work fine under other CFWs and don't have any corruption.

bananapi761, 2018-11-28:
Yes, I'm using the all the files from the 0.7.5 release.

fincs, 2018-11-28:
How did you acquire the software? If they're digital, can you redownload them from the eShop on a fresh SD card?

fennectech, 2018-11-28:
They cant redownload anthing on 6.1.0

ZachyCatGames, 2018-11-29:
Do the games work on stock firmware?

bananapi761, 2018-11-29:
Yes, and on other CFWs.

tiliarou, 2018-12-01:
> Yes, and on other CFWs.

You are probably not using patches properly, which BTW are not supported officially by atmo (AFAIK) as it can enable piracy...

wicksand420, 2018-12-04:
One of the games I found that does this is coloring book. I has the profile select after game boot, and it crashes on atmosphere right before the profile select screen and takes you back to home menu. ReiNX does get past it.

RaldS05, 2018-12-10:
It also happend to me, games are not launching after the user login mode. Games worked normal when I i use the https://github.com/AtlasNX/Kosmos release, run the hektake payload and run Atmosphere from hektake.

SciresM, 2018-12-10:
@RaldS05 does the game launch on Stock?

bananapi761, 2018-12-10:
This seems to be fixed after 0.8, I also had sigpatches but didn't change them before or after updating.

SciresM, 2018-12-11:
I'm going to tentatively mark this as resolved, mostly because I can't reproduce despite lots of effort, and think is probably warez related. 

If someone has a non-warez (officially-bought content) case where this issue reproduces and can perform tests for me, reach out to me on discord.

closed 2018-12-11 by SciresM