Atmosphere-NX/Atmosphere #278: Failed to enable SMMU!

RaneyDay5, 2018-11-30:
## Bug Report
cant boot
### What's the issue you encountered?
whenever i inject the payload it errors with 
[ERROR] Fatal error: [SMMU]: Failed to enable SMMU!
### How can the issue be reproduced?
injecting the payload
### Crash report?

(If a crash report was created under /atmosphere/crash_reports/, please upload it to 
[gist](https://gist.github.com/) and paste the link here.)

### Environment?
Tried fusee and hekate 6.2, Atmosphère version 0.8 
- What bootloader (fusee, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: offical
- Do you have additional kips you're loading:
- Additional info about your environment:

### Additional context?

TheProvider0, 2018-11-30:
The patches found in the latest sdfiles atomsphere folder got me to boot after getting the same error. Just remember to drag the "exefs_patches" folder itself into "atmosphere." If that folder already exists, then delete it and replace it with this one.

RaneyDay5, 2018-11-30:
i got it to work after several formats. Now none of my homebrew apps are showing up in the launcher

TheProvider0, 2018-11-30:
That's another problem to open up, you can close this one since you managed to get it to boot, and remember to always try looking into the discussion thread: https://gbatemp.net/threads/atmosphere-0-8-released-6-2-0-working.524769/page-5#post-8405810

RaneyDay5, 2018-11-30:
okay thanks!

closed 2018-11-30 by RaneyDay5