Atmosphere-NX/Atmosphere #279: Build error

urherenow, 2018-11-30:
## Bug Report
Unable to complete build of Stratosphere. Fails on fatal_task_screen.hpp
### What's the issue you encountered?
/home/urherenow/projects/Atmosphere/stratosphere/fatal/source/fatal_task_screen.hpp:27:9: error: 'NWindow' does not name a type
         NWindow win;
         ^~~~~~~
/home/urherenow/projects/Atmosphere/stratosphere/fatal/source/fatal_task_screen.hpp:28:9: error: 'Framebuffer' does not name a type; did you mean 'setbuffer'?
         Framebuffer fb;
         ^~~~~~~~~~~
         setbuffer

### How can the issue be reproduced?
update all libraries, clone current Atmosphere, and type "make"
### Crash report?
N/A
(If a crash report was created under /atmosphere/crash_reports/, please upload it to 
[gist](https://gist.github.com/) and paste the link here.)

### Environment?
Ubuntu with latest devkit and libnx
- What bootloader (fusee, hekate, etc) was Atmosphère launched by: N/A
- Official release or unofficial build: Official 0.8 source
- Do you have additional kips you're loading: N/A
- Additional info about your environment:

### Additional context?

RaldS05, 2018-11-30:
Having that same build error, I have updated devkit to the latest version. The last version can be compiled with out any problem.

make[2]: Leaving directory '/home/ralds05/SRC-Codes/Atmosphere/stratosphere/creport'
make -C fatal all
make[2]: Entering directory '/home/ralds05/SRC-Codes/Atmosphere/stratosphere/fatal'
fatal_task.cpp
aarch64-none-elf-g++ -MMD -MP -MF /home/ralds05/SRC-Codes/Atmosphere/stratosphere/fatal/build/fatal_task.d -g -Wall -O2 -ffunction-sections -march=armv8-a -mtune=cortex-a57 -mtp=soft -fPIE -DDISABLE_IPC -DATMOSPHERE_GIT_BRANCH=\"master\" -DATMOSPHERE_GIT_REV=\"master-bcdfc53\" -I/home/ralds05/SRC-Codes/Atmosphere/stratosphere/fatal/include -I/home/ralds05/SRC-Codes/Atmosphere/stratosphere/fatal/../../common/include -I/opt/devkitpro/portlibs/switch/include -I/opt/devkitpro/libnx/include -I/home/ralds05/SRC-Codes/Atmosphere/stratosphere/fatal/../libstratosphere/include -I/home/ralds05/SRC-Codes/Atmosphere/stratosphere/fatal/build -D__SWITCH__ `freetype-config --cflags` -fno-rtti -fno-exceptions -std=gnu++17 -c /home/ralds05/SRC-Codes/Atmosphere/stratosphere/fatal/source/fatal_task.cpp -o fatal_task.o 
In file included from /home/ralds05/SRC-Codes/Atmosphere/stratosphere/fatal/source/fatal_task.cpp:22:
/home/ralds05/SRC-Codes/Atmosphere/stratosphere/fatal/source/fatal_task_screen.hpp:27:9: error: 'NWindow' does not name a type
         NWindow win;
         ^~~~~~~
/home/ralds05/SRC-Codes/Atmosphere/stratosphere/fatal/source/fatal_task_screen.hpp:28:9: error: 'Framebuffer' does not name a type; did you mean 'setbuffer'?
         Framebuffer fb;
         ^~~~~~~~~~~
         setbuffer
make[3]: *** [/opt/devkitpro/devkitA64/base_rules:14: fatal_task.o] Error 1
make[2]: *** [Makefile:125: build] Error 2
make[2]: Leaving directory '/home/ralds05/SRC-Codes/Atmosphere/stratosphere/fatal'
make[1]: *** [Makefile:12: fatal] Error 2
make[1]: Leaving directory '/home/ralds05/SRC-Codes/Atmosphere/stratosphere'
make: *** [Makefile:18: stratosphere] Error 2

RaldS05, 2018-11-30:
[image]

soarqin, 2018-11-30:
Do not use libnx from devkitpro, just build libnx from this branch manually: https://github.com/Atmosphere-NX/libnx/tree/ams_fatal_gpu_merge

urherenow, 2018-11-30:
Dang. Lot's of good stuff in the latest official libnx, but there are a few conflicts with the ams_fatal_gpu_merge fork so it can't be merged as-is. This does fix the build though. And I can't help but think I've had the same type of problem with the same solution several times now. When will I learn? SMH...

Thanks @soarqin !

SciresM, 2018-11-30:
@urherenow building ams right now is kind of jank. There's no single branch you can use. Shoot me a DM on discord if you want help building it.

closed 2018-11-30 by SciresM