Atmosphere-NX/Atmosphere #280: Extend BCT.ini to allow management of bundled kips (fs_mitm, etc.)

RShadowhand, 2018-11-30:
## Feature Request

### What feature are you suggesting?

BCT.ini currently doesn't do a lot of useful things, at least not for the end user.
Can we get an extension to this file to allow management of bundled KIPs in stage2?

SciresM, 2018-11-30:
@RShadowhand this sounds strictly less convenient than the current "use all the kips in the kips directory" method.

closed 2018-11-30 by SciresM

RShadowhand, 2018-11-30:
I'm talking about the bundled kips in fusee-secondary.bin.

SciresM, 2018-11-30:
@RShadowhand what specifically do you mean by "management"?

RShadowhand, 2018-11-30:
A way to disable bundled kips, basically. I've been looking for a way to disable fs_mitm globally (not per title), and have found no way other than basically compiling the source with a bool set to false.

SciresM, 2018-11-30:
@RShadowhand Not an option. Atmosphere's KIPs have strong dependencies on one another, and the system will fail to boot if one is not present. 

Why do you not want `fs_mitm` present?

RShadowhand, 2018-11-30:
I'm trying to use ldn_mitm for LAN play through internet. It's failing for certain titles (that work with hekate). I **think** fs_mitm and ldn_mitm are having an argument. Wanted to confirm my suspicions by disabling the bundled fs_mitm.

SciresM, 2018-11-30:
@RShadowhand The two shouldn't be able to "have an argument" -- by design `fs.mitm` is transparent. It's possible that the system is e.g. running low on sysmodule memory, though.

Anyway, the "disable select embedded kips" thing is definitely not an option. In particular, `pm` will fail to boot if `fs.mitm` isn't present.

RShadowhand, 2018-11-30:
Yeah, it's not running out of memory, as the game runs, and has no issues. It just gets stuck on a black screen when "local wireless" is selected. 
It works fine on hekate in 6.1, but not in 6.2 with AMS 0.8.0.
I'm not entirely sure if it's AMS that's causing the issue, or the kip itself.
As for this request not being an option, I understand.

SciresM, 2018-11-30:
Yeah, sorry...

If you want help disabling it the hard way, contact me on Discord. I can do my best to walk you through the process of removing PM's dependency on `fs.mitm`...

RShadowhand, 2018-11-30:
I'll wait for a few more days, as ldn_mitm is under very active development. Maybe there'll be an improvement. If not, I'll contact you through Discord, see if we can find a way out of this.

Thank you for your time.