Atmosphere-NX/Atmosphere #281: Ams version not displayed with exfat sd card

jul2003, 2018-11-30:
## Bug Report

### What's the issue you encountered?
When my sd card is formatted exfat ams version is not displayed on system menu.
If i format it fat32 it is displayed

### How can the issue be reproduced?
format the sd with exfat and put ams file on it and run ams

### Crash report?
no

(If a crash report was created under /atmosphere/crash_reports/, please upload it to 
[gist](https://gist.github.com/) and paste the link here.)

### Environment?

- What bootloader (fusee, hekate, etc) was Atmosphère launched by: fusee-primary.bin
- Official release or unofficial build: official
- Do you have additional kips you're loading: no
- Additional info about your environment: firmware 4.1.0 update to support exfat with choixdujournx

### Additional context?

SciresM, 2018-11-30:
You don't actually have exfat firmware installed. Install exfat firmware.

closed 2018-11-30 by SciresM

jul2003, 2018-11-30:
what do you mean ?
if i run ofw and i format the sd card on the switch it's formatted exfat
and it's a 128g card that was not recognized before i update my 4.1.0 firmware to 4.1.0 exfat using choixdujour