Atmosphere-NX/Atmosphere #284: Detect if an installed title is actually corrupt or not / bypass check.

Serfrost, 2018-12-01:
### Why would this feature be useful?

When Switching from other "CFW" aka Repacks, often some titles will be reported to be corrupt or have corrupt save data.  Horizon initiates a data check, and if it cannot find a solution it will say to redownload, even if the titles worked with another CFW configuration (aka, patches or other.)  It would be nice to bypass this check and hopefully have it display a human-legible deduction of what the problem might be.

It would also be nice if the [ ! ] next to a "Problem Title" is omitted after if the user cancels the check, so users may attempt to launch the software again without having to completely reinstall it from scratch, (after deleting it from the Memory Management console in System Settings.)

SciresM, 2018-12-01:
Please don't seek support for warez here.

closed 2018-12-01 by SciresM

Serfrost, 2018-12-02:
> 
> 
> Please don't seek support for warez here.

Even a "This NSP is invalid." message would have worked, really.

Serfrost, 2018-12-02:
> 
> 
> You wont get that here. We dont support warez

... I can read.

fennectech, 2018-12-02:
Then dont continue asking for something after you’ve been told it wont
happen.

Serfrost, 2018-12-06:
> 
> 
> @Serfrost `some titles will be reported to be corrupt or have corrupt save data.`
> 
> The circumstances under which this occurs are only when one has installed unofficial content. Hence: please do not seek support for warez here.
> 
> This is not an issue that occurs for legitimate content.

I appreciate the further response, SciresM.  I'll keep this in mind in the future, thank you.