Atmosphere-NX/Atmosphere #288: [6.2.0] Atmosphere can't boot, crashes after (Nintendo) logo.

GiovanH, 2018-12-03:
## Bug Report

### What's the issue you encountered?

Atmosphere can't boot on 6.2.0.

Sequence of events:
1. I load the 0.8.1 atmosphere primary payload via fusee
2. Atmosphere gives the friendly message and then loads fusee-secondary.bin
3. Fusee-secondary gives log information that it is beginning NXBoot and rebuilding package 2
4. Fusee-secondary shows several patched kips, as it normally would during a successful boot.
5. The Nintendo logo fades in and out, as it would normally during a successful boot
6. The Switch logo never loads. 
7. At this point, if connected to the computer, the computer loses connection to the switch, indicating some change in power state.

### How can the issue be reproduced?

Issue occurs consistently on my console.

### Crash report?

(If a crash report was created under /atmosphere/crash_reports/, please upload it to 
[gist](https://gist.github.com/) and paste the link here.)

Unfortunately, no crash reports were generated.

### Environment?

- What bootloader (fusee, hekate, etc) was Atmosphère launched by:
Originally loaded the atmosphere chainloader (fusee-primary) via Hekate, but injecting the fusee-primary.bin had the same result. 
- Official release or unofficial build:
Running files from official build 0.8.1
- Do you have additional kips you're loading:
Originally loaded with SysFTPD, disabling sysftpd (moved to different folder) had no effect.
I first ran it with a theme as a LayeredFS patch, but deleting the theme had no effect.
- Additional info about your environment:
Upgraded from 6.1.0 to 6.2.0 using ChoiDejour.

The following environment produced the result:
- Smashed the fusee-primary bin directly via fusee
- SD card with no custom sysmodules or menu themes

### Additional context?

Upgraded from 6.1.0 (with 6.2.0 fuses) to 6.2.0 using ChoiDejour.
**Stock firmware boots correctly.**

GiovanH, 2018-12-03:
Update: This is not exactly true. If left alone for a *long* time (~10 minutes), the switch will occasionally show the Switch logo, boot for another few minutes, and then crash with 2162-0002.

ZachyCatGames, 2018-12-03:
Did you have an exfat card and if so did you install the exfat version of 6.2.0

GiovanH, 2018-12-04:
> This sounds to me like the SD card is failing to mount. Future versions of AMS will have [bbed781](https://github.com/Atmosphere-NX/Atmosphere/commit/bbed78149c7a176a3687dc081f6a34063a603f8e), which will make it take less time to get to the sad state you're in.
> 
> But, yeah, format your SD card as FAT32, or else manually install exFAT firm by fucking with NAND. I'd say use choidujournx, but in order for HBL to load the SD card needs to be readable, which isn't the case.

This was exactly the case. I was able to boot into full atmosphere by using a second SD card with a bare minimum homebrew setup, and from there installed 6.2.0 ExFat with ChoiDejourNX.

Hopefully others who have this problem will see this and be able fix it quickly. Also, apparently atmosphere loads between the Nintendo logo and the switch logo, so that'll be handy to know for future debugging.

Please mark as resolved.

closed 2018-12-04 by SciresM

AbdullahAl-Hajjar, 2020-03-09:
Just update hekate to the latest version don't delete the bootloader folder rather replace it or you will lose your content in the menu page. if it still doesn't work add the latest patches this should work.
I know this is a bit late however I had the same issue and haven't figure it out till recently.