Atmosphere-NX/Atmosphere #291: LayeredFS breaks w/ HBL takeover

jerbear64, 2018-12-04:
## Bug Report

### What's the issue you encountered?
LayeredFS appears to not function when HBL is taking over the target LayeredFS title.

My loader.ini setup is to launch The Binding of Isaac: Afterbirth+ as usual when no buttons are held, but to boot into hbmenu when R is held, since I've heard that R is a LayeredFS override button anyways and I typically play Isaac more than I use homebrew. 

However, LayeredFS does not apply with this setup, as the soundtrack mod I am using no longer works. It begins to work properly again once the loader.ini ID is changed to a different title.

### How can the issue be reproduced?
* Modify a title with LayeredFS
* Open the title, and ensure the mods are present
* Set the same title as the HBL takeover title in loader.ini, ensuring that loader.ini is set up in a way that doesn't disable LayeredFS when the game is launched
* Re-open the title properly, and note that the mods no longer function

### Crash report?

N/A

### Environment?

- What bootloader (fusee, hekate, etc) was Atmosphère launched by:
fusee-primary, chainloaded via Hekate.

- Official release or unofficial build:
Official build, 0.8.1.

- Do you have additional kips you're loading:
None.

- Additional info about your environment:
I am using exFAT, but I don't believe it's relevant in this scenario. Listing anyways, just in case.

### Additional context?

N/A

SciresM, 2018-12-04:
This isn't a bug. It's a relic of your turning override off-by-default.

When you set override_key=R, you're telling loader and fs.mitm "Do not override contents, unless R is held."

Thus, fs.mitm will not apply mods unless you hold R.

closed 2018-12-04 by SciresM

jerbear64, 2018-12-04:
> When you set override_key=R, you're telling loader and fs.mitm "Do not override contents, unless R is held."

Does this mean that it's impossible at this time to use LayeredFS on a takeover title? Because you also said
> I would either change your configuration button in loader.ini

which seems to imply something different, unless I'm misreading it.
I did change override_key to L, not sure why that would do anything but "I would either change your configuration button in loader.ini" implied that it did, and unsurprisingly it had no effect.

SciresM, 2018-12-04:
@jerbear64 I guess you're right. In which case, yes.

I don't anticipate this really changing, as having a single button to control content override globally is pretty nice...

Flumpster, 2018-12-04:
Is this a possibility?

[config]
hbl_tid=010000000000100D
hbl_path=atmosphere/hbl.nsp
hbl_override_key=!L  #only used if added to loader.ini. Fallback is override_key.
override_key=!R