Atmosphere-NX/Atmosphere #301: Request: LayeredFS via live IPS patching

DioVento, 2018-12-16:
## Feature Request

### What feature are you suggesting?
The ability for Atmosphere's LayeredFS module to function with IPS files instead of only with actual system files.
### Why would this feature be useful?
LayeredFS is an amazing tool for loading mods, but sites such as GBATemp have changed their policy on modded game files: No game files, modded or not, can be distributed. All mods are only allowed if distributed as patches. A mod like my [Polished Pikachu & Enhanced Eevee](https://diovento.wordpress.com/lgpplgee/) would require **527** IPS patches to be created and applied. 
Distributing the mod as IPS files would actually be trivial for me, but then my end users would then have a much more complicated process to actually play the mod: They'd need to dump their entire game (potentially requiring an exFAT SD card), extract it (using dumped keys to do so), and then use a complicated batch script (that I'd need to write, and would only work on Windows because I don't know how to make scripts for Linux or OSX) to individually IPS patch each altered file. This drastically raises the barrier to entry for playing a mod, since the current distribution method is as simple as "drag and drop to SD card while using Atmosphere".
If Atmosphere's LayeredFS supported live patching via IPS files, I could just distribute a folder full of IPS files that would work over a game, and it would eliminate the "user convenience versus guaranteed legality" dilemma that currently exists in distributing mods.

SciresM, 2018-12-17:
Yeah, sure.

AnalogMan151, 2019-12-11:
Did this get implemented? I don't remember reading anything about it.

SciresM, 2019-12-11:
Ah, no, it didn't -- it also kind of can't really be implemented for technical/performance reasons. It's been discussed a few times in RS. Talk to me later for more details. Closing for now until a time comes that a technical solution is viable.

closed 2019-12-11 by SciresM