Atmosphere-NX/Atmosphere #305: fusee: -Werror and -Wall not a good combination

misson20000, 2018-12-21:
I feel like this discussion has happened somewhere before, but I'm having trouble finding it. The Makefiles for fusee-primary and fusee-secondary specify both `-Werror` and `-Wall`, which can break your builds when a compiler updates adds warnings to your code that weren't recognized before. Removing `-Werror` from fusee-primary and fusee-secondary Makefiles just to get it building is one of those changes that's constantly dirtying my working directory.

profi200, 2018-12-22:
I would leave that on for at least release builds. If there are new warnings there is possibly something broken that went unnoticed before.

misson20000, 2018-12-27:
In my experience, new warnings typically indicate code-smells, but not that anything is actually broken. There are a lot of warnings that mean that your code actually is broken (uninitialized variables, missing return statements, etc.), and I strongly recommend that those warnings be `-Werror`'d, but `-Wall` also pulls in a lot of warnings that really don't matter as much.

The sdmmc driver currently uses some type punning that triggers some warnings. It should probably be cleaned up, but it doesn't break the code. I've had compilation fail because the compiler thought there was a possibility for `snprintf` to truncate the result. Yeah, it's an issue, yeah it should be fixed, yeah you shouldn't release with warnings, but I really don't care if someone's `snprintf` might truncate when I'm just trying to build fusee.

misson20000, 2019-03-21:
Update March 2019: `-Wall -Werror` not looking like any better of a combination than it did four months ago and now sept builds fail too because of `-Werror=strict-aliasing` (using arm-none-eabi-gcc from official arch repos).

fincs, 2019-03-21:
Atmosphère is built using devkitARM and devkitA64, not some other random arm-none-eabi-gcc toolchain. I don't think anyone else has this problem. I've checked myself and sept builds with no errors or warnings.

TuxSH, 2019-03-21:
`-Wall` is not `-Wextra`.

Also fusée will not work w/o dkA, dunno about sept.

The strict aliasing warning could come from the very latest gcc or system default flags. I'm interested in what's causing it.

fincs, 2019-03-21:
For the record:

```
arm-none-eabi-gcc (devkitARM release 52) 8.3.0
```

SciresM, 2019-03-21:
@misson20000 closing, can't reproduce build failure (and I want Wall/Werror on for fusee/sept).

closed 2019-03-21 by SciresM

misson20000, 2019-03-21:
@fincs hmm, yeah, I'd thought that the Makefile was using system `arm-none-eabi-gcc` from Arch repos because which `arm-none-eabi-gcc` didn't return a dkA path, but the Makefiles are apparently more clever than I (and `which`) am and it turns out I was using a severely outdated version of devkitARM that I'd installed before devkitPro had pacman repos.

I still personally think that `-Wall -Werror` is bad taste since it's liable to cause builds to fail whenever a toolchain update adds new warnings, but this did turn out to be a *me* problem as opposed to an Atmsophere problem.

fincs, 2019-03-21:
@misson20000 I still personally think that silencing compiler warnings is bad taste, and people should fix their code as compilers get better and are able to detect more and more issues within their code. Enabling `-Wall -Werror` forces you to do the right thing.