Atmosphere-NX/Atmosphere #308: Random gray screen hw crash

cucholix, 2018-12-23:
Today was creating a Mii while staying in atmosphere cfw, and the OS freezes, no button input respond, when I press Home button the screen went gray and get stuck... Happened twice now, previously was sending a screenshot to Twitter when happened. 

Is there anyway to get the system back to normal? Had to completely turn it off, and inject the payload inject all over again...

My Switch is 6.2.0 no homebrew launcher installed, only using atmosphere to patch some game files (game mod). 

Injected fuse primary bin file using TegraRCM GUI

Using:
https://github.com/Atmosphere-NX/Atmosphere/releases/tag/0.8.2

ZachyCatGames, 2018-12-23:
I've also had this happen a few times, it'll just randomly freeze sometimes when I get to a loading screen. It started happening to me after updating to 6.2.0 when Atmosphere 0.8.0 came out, although its only happened 2-3 times for me

fincs, 2019-01-22:
Can reproduce this on 6.2; it has happened to me twice today. Seems to be related to applet transitions somehow (maybe qlaunch crashes?)

fennectech, 2019-01-22:
I have a simulacra hang when i leave YouTube open overnight.   But it doesnt do the gray screen. The display just shows the same image and the LCD doesn’t come up when i pull it off the dock.  I have to do a 15 second power button and reboot to get back to normal

cucholix, 2019-01-22:
It seems the “grey screen” depends on the system theme (dark in my case)
If you guys can confirm it?

GRAnimated, 2019-01-22:
Definitely a qlaunch issue. You guys running any custom themes?

cucholix, 2019-01-22:
No custom theme here

fincs, 2019-01-22:
I have never attempted corrupting qlaunch with unofficial modifications.

fennectech, 2019-01-22:
No custom themes here.  But if i launch without creport i dont get a hang and instaid a normal “the software closed because an error occored” that kicks me to the home screen.

cucholix, 2019-01-24:
Got the first crash during gameplay (Dark Souls Remastered) without the “The software closed because an error occored” message. The only difference this time I was playing in handheld mode, without internet connection, I use to play this game in TV mode with Switch Pro and internet connection enabled.

fennectech, 2019-01-25:
My varient has vanished with 0.8.3   Do you guys still get yours?

jorgesd4, 2019-01-25:
Hi. I had this issue in the version 0.8.2 using 6.2 in my console. Just one time it happened. Now i got this same grey screen after updating to 0.8.3. Using atmosphere using dark themes and no custom themes. It just happens when i was entering the album menu and i press R to avoid hbmenu.

closed 2019-05-22 by SciresM