Atmosphere-NX/Atmosphere #312: BotW not launching on 0.8.2 & 3.0.2

kuritonasu, 2018-12-28:
## Bug Report

### What's the issue you encountered?
BotW doesn't start from its gc on Atmosphere 0.8.2 . 1-2 Switch with its gc works without issues. eShop bought games work without issues. Booting to stock and trying, BotW works without issues. I'm on 3.0.2 since I bought the switch over a year ago. gc isn't disabled.

### How can the issue be reproduced?
Boot to Atmosphere. Have BotW cart inserted (or insert). Launch BotW. The error "The software was closed because an error occurred" is displayed.

### Crash report?
https://gist.github.com/kuritonasu/aed7249a91fd7185a3542498b77c86ef

### Environment?
- What bootloader (fusee, hekate, etc) was Atmosphère launched by:
+ Hekate 4.5. Also tried with fusee-primary bundled with 0.8.2
- Official release or unofficial build:
+ Official. Tried both Kosmos and Atmosphere only on clean SD
- Do you have additional kips you're loading:
+ No
- Additional info about your environment:
+ Nothing else really. I tried restoring my clean nand backup just in case; no change, issue still persists.

### Additional context?

kuritonasu, 2018-12-28:
BotW is v1.3.1

kuritonasu, 2019-01-05:
@tiliarou I have checked, yes. I'm guessing that it wouldn't launch in stock either if there was data corruption.

SciresM, 2019-01-25:
Closing due to lack of info, re-open new issue if still relevant.

closed 2019-01-25 by SciresM