Atmosphere-NX/Atmosphere #314: Actually block auto updates

nbalonso, 2019-01-01:
## Feature Request
Respect the user's choice of not auto updating

### What feature are you suggesting?
Prevent Horizon from downloading or writing the updates in the background. If the update is already downloaded and they user has the auto update turned off, delete it.

Hopefully blocking before the request reaches N's servers to leave no trace. If not possible, prevent those files from being written to disk

### Why would this feature be useful?
Currently Horizon downloads the updates in the background no matter of what the user has set in settings. Then you get nagged to update on each game launch. This is bad behaviour by N

Tested on 5.1.0+Atmosphère 0.8.2, auto update turned off, been on wifi for ~1min --> "An update is ready", pressed A twice while opening a game... welcome to 6.2.0.

Thankfully I had autoRCM enabled and fuses were harmed

nbalonso, 2019-01-01:
I suppose what I'm asking for is to be able to kill and prevent from running the ns:su service by toggling the setting

Bonus points for deleting the downloaded update data

ireece1, 2019-02-16:
This is a must. 

When trying to load a game, without even thinking about it i accidentally clicked update on the nag prompt and it took me from 6.2.0 to 7.0.0.

No more Atmosphere until they get 7.0.0 figured out.

whita019, 2019-02-20:
Please we need this

ValliantStorme, 2019-02-21:
If you need to block updates, use 90dns, and enter Maintenance Mode at least once afterwards.

jonirrings, 2019-02-21:
i deployed 90dns `dnsmasq config` to my openwrt. so all ninty communication are blocked

ccfman2004, 2019-02-21:
90dns must be deployed on a per network instance so any Pre-loaded Nintendo WiFi's will be excluded so a way to disable system update traffic would be great.

fennectech, 2019-02-21:
Why not allow users to specify URLS blocked via a config file?    So that we don’t have to use dnsmasq on a per network basis to partially block Nintendo domains?

SciresM, 2019-05-23:
Closing this issue since no work is being done on this, however given a concrete proposal for *how* to achieve this, I'm theoretically open to it in the future.

closed 2019-05-23 by SciresM