Atmosphere-NX/Atmosphere #318: Request: Do research into what causes bans

TheCoder3156, 2019-01-06:
## Feature Request

Do research into what causes bans.

### What feature are you suggesting?

Do through investigations on what causes bans as to prevent them (and stupid drama). Side note: I think the Nintendo sdk might implement cfw detection into games themselves.

### Why would this feature be useful?

It would prevent drama among people and people can still get game updates without doing illegal downloading.

DavidBuchanan314, 2019-01-06:
Will you be donating the sacrificial ban test switches? :P

SciresM, 2019-01-06:
Not a priority at the moment -- the only reported cases of bans have been records of unofficial content making their way to N, or else other misbehavior getting sent to them. Better, I'd think, to continue improving anti-telemetry, and evaluate only then.

closed 2019-01-06 by SciresM

TheCoder3156, 2019-01-06:
Well that's kinda what I meant. Just improve the anti ban modules

SciresM, 2019-01-06:
If you meant "improve anti-telemetry", why was your issue titled "perform research"?

TheCoder3156, 2019-01-06:
I meant to see what causes the anti-telemetry to be sent.

TheCoder3156, 2019-01-06:
Also mind if I make other feature requests? I don't want to be rude or spammy.