Atmosphere-NX/Atmosphere #322: Hang when playing "EVERYTHING"

fennectech, 2019-01-11:
## Bug Report
System lockup  (not fatal but a hang where entire console has frozen) 
### What's the issue you encountered?
If you transform into a kit fox in the game "everything" in this situation (save attached) the system will crash.   
### How can the issue be reproduced?
Simply restore this save and transform into a kit fox (in favorites) and the console will freeze
### Crash report?
N/A   No report generated as fatal/creport is not invoked.
(If a crash report was created under /atmosphere/crash_reports/, please upload it to 
[gist](https://gist.github.com/) and paste the link here.)

### Environment?
Stock Atmosphere 0.8.2  No other kips
- What bootloader (fusee, hekate, etc) was Atmosphère launched by: fusee-primary launched by hekate  launched by sxos     Also happens when launching by fusee-primary.bin
- Official release or unofficial build: Official build.
- Do you have additional kips you're loading:
happens no mater what set of kips are loaded  including none  
- Additional info about your environment:
None.
### Additional context?
Here is the save file that triggers the crash.    An unaltered save created via normal gameplay.

https://www.dropbox.com/s/ffeplduqk83qf2m/20190111-001818%20FennecTECH.rar?dl=0

fennectech, 2019-01-11:
Additional discovery   Without @SciresM's modified creport loaded i get a standard "the software was closed because an error occored.

ThatNerdyPikachu, 2019-01-13:
> Additional discovery Without @SciresM's modified creport loaded i get a standard "the software was closed because an error occored. Something appears to be causing creport to hang.

But...but...

``without it loaded`` -> ``standard error``

...

p.s. you mean fatal

fennectech, 2019-01-13:
NO.  I’m talking about creport.     With creport loaded it hangs.  Without creport loaded i get a standard “the software was closed because an error occored” that throws me out to the home screen.

fennectech, 2019-01-25:
Whoo  Actual stability!   This was fixed in 0.8.3 so i will close this issue.   Thank you :)

closed 2019-01-25 by fennectech