Atmosphere-NX/Atmosphere #330: Reboot to Payload crashes when attempting to reboot

Hydr8gon, 2019-01-25:
## Bug Report

### What's the issue you encountered?

When attempting to reboot to payload with the example homebrew included with Atmosphère 0.8.3, the program crashes when pressing the - button to reboot.

### How can the issue be reproduced?

Run the Reboot to Payload homebrew with hbmenu (can be launched by overriding an installed app or a system applet, both crash).

### Crash Report

https://gist.github.com/Hydr8gon/675e6ccaaf6ab8059fadd61e1a90bd6f

### System Firmware Version

6.2.0

### Environment?

Launched with latest fusèe payload, official Atmosphère 0.8.3 release. No additional kips loaded.

### Additional context?

Rebooting to payload works fine from the fatal crash screen, just not from the sample homebrew.

JamePeng, 2019-01-25:
disable the reinx_toolkit debug mode, it will make the homebrew software crash.

SciresM, 2019-01-25:
Update your HBL -- it requires HBL 2.1.0.

closed 2019-01-25 by SciresM

Hydr8gon, 2019-01-25:
Ah whoops, my bad. I forgot hbloader and hbmenu were separate things... Thanks.