Atmosphere-NX/Atmosphere #331: Offer a way to still run games with fs_mitm mods while title_id=app is used

thedax, 2019-01-25:
## Feature Request

### What feature are you suggesting?
#### Overview:
If `title_id=app` is used, it's impossible to also play games with fs_mitm mods, since the override key will only launch the HBL (if override_key is R for example) or the game without mods (with !R override key). I propose that an additional key be added to temporarily ignore `title_id=app` so that titles with fs_mitm mods can still be used.

#### Smaller Details: 
N/A

#### Nature of Request:  
Change/Addition (falls under both)

### What component do you feel this would best fit within?
Stratosphere: loader

### Why would this feature be useful?

As it stands currently, the user has to manually edit loader.ini back to a title id, put it back on the console, and reboot before it takes effect (a bug perhaps? IIRC loader is supposed to be able to do this on the fly).

SciresM, 2019-01-25:
Duplicate of #329

closed 2019-01-25 by SciresM

thedax, 2019-01-25:
I disagree, but alright.