Atmosphere-NX/Atmosphere #332: BotW not launching on ams 0.8.2/0.8.3, horizon 3.0.2

kuritonasu, 2019-01-25:
## Bug Report

### What's the issue you encountered?
BotW doesn't start from its gc on Atmosphere 0.8.2 & 0.8.3 . 1-2 Switch with its gc works without issues. eShop bought games work without issues. Booting to stock and trying, BotW works without issues. I'm on 3.0.2 since I bought the switch over a year ago. gc isn't disabled.

### How can the issue be reproduced?
Boot to Atmosphere. Have BotW cart inserted (or insert). Launch BotW. The error "The software was closed because an error occurred" is displayed. If I reboot to stock, BotW launches fine.

### Crash report?
https://gist.github.com/kuritonasu/aed7249a91fd7185a3542498b77c86ef (updated for 0.8.3)

### Environment?
- What bootloader (fusee, hekate, etc) was Atmosphère launched by:
Hekate 4.6. Also tried with fusee-primary bundled with 0.8.3
- Official release or unofficial build:
Official. Tried both Kosmos and Atmosphere only on clean SD
- Do you have additional kips you're loading:
No
- Additional info about your environment:
Nothing else really. I tried restoring my clean nand backup just in case; no change, issue still persists.

### Additional context?
Previous bug report (#312) was closed because of lack of information. I've provided all fields required for bug reports, let me know what further information you need.

SciresM, 2019-01-25:
@kuritonasu Are you available to try debugging with my on discord? What version of BotW?

kuritonasu, 2019-01-25:
> What version of BotW?
It's v1.3.1

SciresM, 2019-01-25:
Fixed by da68d02.

closed 2019-01-25 by SciresM

Burgos91, 2019-01-28:
I have the same issue but with a nsp file of botw i tried atmosphere 8.2 and it work fine the issue seem only appear on 8.3 update