Atmosphere-NX/Atmosphere #334: Can't boot into atmosphere from a cold boot.

fOmey, 2019-01-25:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

When cold booting into atmosphere, it usually fails and throws me back into RCM mode. On the second attempt it _usually_ works.. however this morning a total of 3 attempts until a successful boot.

### How can the issue be reproduced?

Completely shut the switch down and attempt to boot into atmosphere.

### Crash Report

N/A

### System Firmware Version

6.2.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
Fusee
- Official release or unofficial build: 
Official: 0.8.3
- Do you have additional kips or sysmodules you're loading:
No
- Homebrew software installed:
Yes, a single piece of homebrew: retroarch.

### Additional context?

My switch is pretty bare bones.. there really isn't much room for outside error.

SciresM, 2019-01-26:
So, the issue is that sometimes booting fails? But you have managed to boot successfully?

fOmey, 2019-01-26:
Correct, usally takes two boots for a successful boot.

I can't recall a time I have been able to power down my switch and boot into my atmosphere first try.. almost always takes two attempts.

The first attempt will run through the boot up sequence and when you would expect to see the Nintendo logo it returns to rcm mode and I'll have to resend the payload.

misson20000, 2019-01-26:
I think I have this happen too, although I never noticed that it happens after power offs.

SciresM, 2019-01-26:
@fOmey @misson20000 Some system components won't work right if a full reboot is not done -- e.g. a PMC reboot. A watchdog reboot is insufficient, for example. This could mean that depending on how you boot into ams, the first attempt at booting fusee might fail. I've mitigated this somewhat in ea02f389 and 3f6325c3, and fusee now reboots into itself instead of into RCM (so if you get a failure, you won't need to re-launch an RCM payload or anything).

I think this is basically closeable, since the problem isn't fundamentally fixable? Will leave open for 24 hours, just in case.

closed 2019-01-26 by fOmey