Atmosphere-NX/Atmosphere #335: 7.0.0

melsbacksfriend, 2019-01-31:
I heard both that 7.0 broke cfw and that it has been cracked.  Because it's been cracked, you should add support for 7.0.

closed 2019-01-31 by SciresM

ThatNerdyPikachu, 2019-01-31:
As I'm not sure if this is a troll or not, please, just wait for an official release instead of begging. This is the second time you've done this,

Sovnit, 2019-01-31:
You don't just "add support". Show some respect for his hard work and be thankful someone else can do what you can't.

ZachyCatGames, 2019-02-01:
If it was simple it would have already been done, but it’s not simple. It’s being worked on, and if you want more information about its development, both SciresM’s twitter and the ReSwitched discord server are both great places to watch