Atmosphere-NX/Atmosphere #336: Add a passcode lock to Atmosphère

TTTechnoPro, 2019-01-31:
## Feature Request by TTTechnoPro

### What feature are you suggesting?
A PIN to lock the Switch. 

#### Overview:
This Pin can be digits (like a passcode) or specific button mapping. (ex: ZL + L + ZR + R could be my PIN)

#### Smaller Details: 
Replace the "Press any button 3 times" with a PIN Screen. Upon waking up from sleep mode, press any button to activate the lock and enter PIN to access home screen. 

Possibly lock anything you want, sort of like the iOS Jailbreak tweak "BioProtect"
- Lock specific games and require pin to access
- Lock Album app (prevent minors from messing with HomeBrew)
- etc, the list goes on.

Maybe 4-6 Digits
A message notifying your switch is disabled for "x" minutes (like how most smartphones do it)

#### Nature of Request: 
- Addition of the "Please enter your PIN" message
- Change: N/A
- Removal of the "Press any button 3 times" message

### What component do you feel this would best fit within?
This would best fit within Troposphère but could apply anywhere

### Why would this feature be useful?

This feature would be useful to prevent thieves from stealing your switch and using it. As of now, without Parental Control, any one can steal your switch, erase SD and NAND, and use it. 
A simple PIN could prevent others from stealing it.

Mitsukiii, 2019-01-31:
> This feature would be useful to prevent thieves from stealing your switch and using it

Boot into OFW and you don't have a Pin Lock?

TTTechnoPro, 2019-01-31:
Aww, I forgot about that. 

Require AutoRCM and require PIN to turn off? 
Possibly implement this until an untethered jailbreak is achieved.

shchmue, 2019-01-31:
Where is the PIN stored, SD? Remove the SD. Boot Hekate -> stock.

thedax, 2019-01-31:
AutoRCM wouldn't change anything, any payload can be pushed, so the attacker/thief would have full control the minute they turn your console on.

Rickardo987, 2019-01-31:
The pin lock could be mainly for preventing minors from messing with HB components and therefore wouldn’t have to be 100% secure. I don’t think 5 year old me could find a way to disable AutoRCM and boot into OFW. This could also be easially built into a boot loader such as hekate.

TTTechnoPro, 2019-01-31:
> Where is the PIN stored, SD? Remove the SD. Boot Hekate -> stock.

 if PIN is stored on NAND? Would this still work?

> AutoRCM wouldn't change anything, any payload can be pushed, so the attacker/thief would have full control the minute they turn your console on.

yeah that's correct. This seems like an idea meant for unknowledgeable thieves but once a untethered jailbreak comes, it looks like security for all

ZachyCatGames, 2019-01-31:
>once a untethered jailbreak comes

if a untethered jailbreak comes*

SciresM, 2019-01-31:
Nah.

closed 2019-01-31 by SciresM

ThatNerdyPikachu, 2019-01-31:
>> once a untethered jailbreak comes

> if a untethered jailbreak comes*

if an untethered jailbreak comes**