Atmosphere-NX/Atmosphere #338: No load kips/title folder

BadWolf-alfa, 2019-02-03:
Hello, i compiled (succed), a version of Atmosphere, but when i launch installer for games backup, say error on NCA signature, and if i install a custom theme, are same, for maintain the real atmosphere i changed on my own build the folder to ATM on all strings  (and are correct), the settings say ATM 0.8.3 same as real but no work more (installing backups and custom themes on TITLE FOLDER) why no working ?

Any point? thanks in advance and sorry

lookye, 2019-02-03:
Atmosphere does not support piracy. This is why you can not install your "game backups".

BadWolf-alfa, 2019-02-03:
i use the real ATMOSPHERE and i can install "game backups" ;)

and for CUSTOM THEMES? no allowed anyway? ah?

closed 2019-02-03 by fincs