Atmosphere-NX/Atmosphere #341: creport hangs system.

fennectech, 2019-02-06:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

### What's the issue you encountered?
The console hangs when launching cities skylines with modern creport.  however when downgrading to the creport included with 0.7.4 the hang vanishes and i get a "the software was closed because an error occored" 

[ Describe the issue in detail and what you were doing beforehand. ]
Launching various games including "cities skylines" where this hang occors
[ Did you make any changes related to Atmosphère itself? ]
no
[ If so, make sure to include details relating to what exactly you changed. ]</br>
NA
### How can the issue be reproduced?
Launch cities skylines with current update
[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>
open cities skylines with current version of creport.
### Crash Report
crash report was generated with creport 1.1
https://pastebin.com/2WcGcPTH

(https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

6.2.0</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?
standard kosmos with FTPD 
- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
Hekate 
- Official release or unofficial build: 
  - [ Offical release version x.x.x (or) unofficial build ]
Unofficial build:Kosmos
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
FTPD
- Homebrew software installed: [ * ]

### Additional context?
When using the older version (1.1) of creport the hang does not occor   but when using 1.2 of creport it hangs forcing a reboot.
- Additional info about your environment:
- [ Any other information relevant to your issue. ]

thedax, 2019-02-06:
>standard kosmos with FTPD

Does the issue persist with the latest official release (i.e. https://github.com/Atmosphere-NX/Atmosphere/releases/tag/0.8.3)?

fennectech, 2019-02-06:
Yes.

fennectech, 2019-02-06:
Have i done that right?

thecogdoctor, 2019-02-06:
> 
> 
> Have i done that right?

In the majority yes.

>Launching various games including "cities skylines" where this hang occors
>no

Please include the context to the why you said no. It is present within the issue template but it's nonsense without context

> (https://gist.github.com/) and pasting the link here. ]
>
>]
...

Cheers.

fennectech, 2019-03-08:
I’m not sure why i said no. But i removed it.

fennectech, 2019-03-08:
But the issue seems to be fixed for now.  Ill reopen if i come across it again

closed 2019-03-08 by fennectech