Atmosphere-NX/Atmosphere #344: Game card cannot be read (when upgraded to 0.8.3)

PP-theSLAYER, 2019-02-15:
## Bug Report

### What's the issue you encountered?
When I upgrade my Atmosphere to 0.8.3, the Switch boots just fine, but it says Game card cannot be detected.<br>
Downgrading to the version I was using (0.7.5) of Atmosphere allows it to work just fine, problem is I cannot upgrade my Atmosphere. (and in turn, I probably can't update the Switch's FW either)<br>

### How can the issue be reproduced?

1. I deleted the old Atmosphere folder, and fusee bins, and bootloader files (not sure if these are even used; could just be files I had previously) on my SD
2. I paste in the new ones from the zip on this git (including the updated fusee-primary)
3. I also paste over the fusee-primary on my PC (as I'm using TegraRcmSmash to enter Atmosphere)
4. Switch Boots up just fine.
5. If game cart is in the slot, it says cannot be read.

### Crash Report

I don't think there's any associated crash report.<br>

### System Firmware Version

6.1.0 (AMS 0.7.5) [before upgrading Atmosphere]</br>

### Environment?

I launched Atmosphere straight with TegraRcmSmash (no other selection screen, besides KIPs being loaded), so I don't believe I'm using any additional bootloaders.<br>

My present setup has this in the Atmosphere folder: exef_patches > Signature Patches by Rajkosto, however when I update Atmosphere to 0.8.3, I delete the entire folder, so those files were purged in the process. 

Not sure if that affects the _Game card issue_ I have whenever I re-attempt to update Atmosphere. (I could test it, if need be)

### Additional context?

My Switch was on the lowest firmware possible to play Mario Odyssey (I think 3.0.1) before I installed Atmosphere and used CJNX to update the Switch's FW to play Pokemon. 
Why I mentioned this is because to my understanding, if _Game card reader_ wasn't disabled when doing an update on CJNX, it may make it not functional when the Switch itself is downgraded.

I can't remember for sure, but I believe I did this step, and this may be relevant to the issue I have on hand.

pixel-stuck, 2019-02-15:
if your fuses are still on the older version, newer AMS versions have a feature (nogc) which will prevent the firmware on the chip that reads gamecards from being updated. Seeing as you've already updated the firmware on that chip, you can disable nogc in your BCT.ini (see [here](https://github.com/Atmosphere-NX/Atmosphere/blob/master/common/defaults/BCT.ini#L14) for instructions)

PP-theSLAYER, 2019-02-15:
Thanks for the prompt reply. I'll edit the BCT.ini and try again :)

closed 2019-02-15 by PP-theSLAYER