Atmosphere-NX/Atmosphere #349: Hide Thermosphère (emuNAND) in a randomly generated folder

lordelan, 2019-02-19:
While I don't know how Thermosphère is intended to work in the future, there would be the need for it to be stored somewhere, ideally on the mSD card.

To avoid that _a certain company_ is looking for specific folders and files in order to decide whether to pull the ban hammer or not **and** to avoid having them looking [here](https://github.com/Atmosphere-NX/Atmosphere) how that folder would be named, I suggest to implement a random folder name generator, that generates a folder according to the scheme of that certain company and store it inside sd:/Nintendo/

It should also generate a fusee primary payload for booting the emuNAND containing that specific just generated folder name while saving the path to that folder into a *.txt file on the SD so people know the folder name and can put it into a Thermosphère launch option in hekate if they prefer that over fusee primary. Obviously the *.txt file should be deleted then by the user or automatically by some future hekate version.

Obviously while writing the name in the hekate.ini it is still detectable so this would remain risky but using the just generated fusee primary payload should be as safe as we can get.

misson20000, 2019-02-19:
Is this actually a real problem? I don't think anyone's actually gotten banned for having a folder called `atmosphere` on their SD card.

sigboe, 2019-02-19:
If Nintendo wanted to ban people for having an atmosphre folder on the SD card, and the name was random. Then they could just search for the contents.

Security through obscurity is no security at all.

Candunc, 2019-02-19:
If you're that worried about getting banned have two different SD cards and swap them before booting into Atmosphere. And I don't believe Nintendo issues bans based on folder names alone at the moment.

DavidBuchanan314, 2019-02-19:
I don't know what the actual plan is, but I assume the emunand will be in a partition, not a file.

RyuSeisuke1992, 2019-02-19:
That would be bizarre if they also analyzed the SD card. But we should test it.  I create a folder on the SD called "please ban me Nintendo, I am currently using a CFW on my switch". xD

closed 2019-02-19 by SciresM