Atmosphere-NX/Atmosphere #350: Use built-in controller HOME button light

cucholix, 2019-02-20:
Cosmetic fun request, the right Joy Con, and Switch Pro Controller have a built-in light blue ring in the HOME button, it seems Nintendo have them for some use in the future but they are sitting there in both controllers. When connected to PC and using them as Steam controller the light turns on.

It could be cool to have an option to turn on/off this light upon Switch+atmos boot. 

Regards, and thanks for this godlike CFW =)

TheCoder3156, 2019-02-20:
This isn't something that should go into vanilla atmosphere. If you really want this make an issue/pull request under libnx for support to control the light. Then make a sysmodule that make use of the libnx feature when it comes out.

cucholix, 2019-02-20:
> This isn't something that should go into vanilla atmosphere. If you really want this make an issue/pull request under libnx for support to control the light. Then make a sysmodule that make use of the libnx feature when it comes out.

Well, I like the idea of homebrew apps use this feature, but I'm thinking it being available while using retail games, or Switch menu, for example letting the light always on (optional through a config.txt). I open a request on libnx too.

ValliantStorme, 2019-02-21:
That sounds like something a sysmodule would do (a la sys-clk or sys-ftpd)

closed 2019-02-21 by SciresM