Atmosphere-NX/Atmosphere #354: Switch boots to black screen; no splash displayed

sandhoners123, 2019-02-21:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?
My switch just shows a brief line of white text and then goes to a fully black screen
[ Describe the issue in detail and what you were doing beforehand. ]</br>
My switch just shows a brief line of white text and then goes to a fully black screen. I booted my switch using fusee secondary and copied all the .bin files to atmosphere/ on my sd card. I am on 7.0.0. Self built
[ Did you make any changes related to Atmosphère itself? ]</br>
No
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>
Just load fusee-primary.bin normally on 7.0.0
### Crash Report
There are no crash reports
[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version
7.0.0
7.0.0</br>
//eplace X's with system firmware version at time of crash. ]</br>
//[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[// If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. </br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Offical release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
Self compiled
### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]
Fedora Rawhide build system

SciresM, 2019-02-21:
You probably don't have keys to build for 7.0.0 support.

closed 2019-02-21 by SciresM

sandhoners123, 2019-02-21:
How do you get those keys

SciresM, 2019-02-21:
hax