Atmosphere-NX/Atmosphere #355: Freeze and blackscreen when trying to launch game

Skalke, 2019-02-21:
## Bug Report

### What's the issue you encountered?
Critical freeze and blackscreen when trying to launch game _Modern Combat Blackout_
Yes, i did. Added ldn_mitm_v1.1.2.kip and added some titleids:
00FF0000636C6BFF - Overclock Module by m4xw
00FF747765616BFF - supernag patcher for BrowseNX by crc-32
0100000000001000 - qlaunch
0100000000001013 - qlaunch

### How can the issue be reproduced?

Just try to lauch this game in Atmosphere 0.8.3 on 6.2.0 firmware.

### Crash Report

Couldn't create Crash Report Logs and Dump

### System Firmware Version

6.2.0

### Environment?

- Bootloader was Atmosphère launched by:
Hekate -> Fusee-1 -> Fusee-2
- Official release or unofficial build: 
  - 0.8.3 official
 
- Do you have additional kips or sysmodules you're loading: 
ldn_mitm_v1.1.2.kip
- Homebrew software installed:
Titleids:
00FF0000636C6BFF - Overclock Module by m4xw
00FF747765616BFF - supernag patcher for BrowseNX by crc-32
0100000000001000 - qlaunch
0100000000001013 - qlaunch

and  a lot of stuff (i can add it if it needed)

### Additional context?

I think it could be LFS issue...

Tried Kosmos v11.9 with all my mine configs from atmo in sigchk/nosigchk-mode - it's working great.
But in the atmosphere, the game does not want to run.

m4xw, 2019-02-21:
Since you submitted a Issue here, make sure you verify it with _stock clocks_, as OC can cause issues.

Skalke, 2019-02-21:
@m4xw oh...downclock qlauch could cause this issue? when why i don't have problem with kosmos?

Skalke, 2019-02-21:
solved with atmo 0.8.4 lol....configs didnt change

closed 2019-02-21 by Skalke

Grayfoxst, 2020-01-15:
any update on this? im having the same issue