Atmosphere-NX/Atmosphere #357: Bug after loading 8.4 firmware 7.0.0

Burgos91, 2019-02-21:
## Bug Report
All previus games and tinfoil

### What's the issue you encountered?
After loading atmospher 8.4 all my previus installed games got corrupted and tinfoil nor niuds work to install them again.
No change were made

### How can the issue be reproduced?
trying to load a game or installing a game
### Crash Report
N/A

### System Firmware Version

7.0.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
fusee and atmosphere 8.4
- Do you have additional kips or sysmodules you're loading:
no
- Homebrew software installed: [ * ]
yes and runing nicely

### Additional context?
no so far

SciresM, 2019-02-21:
We don't support warez

closed 2019-02-21 by SciresM

powmaya, 2019-02-22:
Weren't you the one who was always playing, datamining and leaking Omega Ruby (and other Pokemon games) on 3ds ahead of release? If that's not piracy I'm not sure what is friend. 

Glass houses and all that.

https://projectpokemon.org/index/news/ultra-sun-and-ultra-moon-datamining-r66/
http://227.149.197.104.bc.googleusercontent.com/2016/11/08/pokemon-sun-and-moon-is-leaked-datamined/