Atmosphere-NX/Atmosphere #358: Open power menu via sleep button

Strat00s, 2019-02-21:
### What feature are you suggesting?
#### Overview:
Kinda self-explanatory. When clicking the sleep button instead of opening the annoying dialog if I really want to sleep the console, simply open the power menu.

### Why would this feature be useful?
A faster way of shutting down the console. Instead of holding the power button for 5 seconds, just click the sleep button and chose power off.
Also very useful to shut down the console when it is docked. You can click power off right from your couch.

misson20000, 2019-02-21:
You mean the buttons on the home menu and overlay menus, not the actual physical button on the console, right?

Sounds to me like it might be a little out of scope for ams, but I like the idea.

Strat00s, 2019-02-21:
> You mean the buttons on the home menu and overlay menus, not the actual physical button on the console, right?
> 
> Sounds to me like it might be a little out of scope for ams, but I like the idea.

Exactly. I should have said sleep icon. The button or whatever on the home screen.

james-d-elliott, 2019-04-14:
I don't know why Nintendo didn't do this themselves, rather annoying having to hold the power button to restart. I think it'd be rather useful for restarting into hekate for removal of microSD for example.

fennectech, 2019-04-14:
Or restarting when connected to TV

SciresM, 2019-04-20:
Yeah, this is a little out of scope due to it involving actual code changes to system applets.

closed 2019-04-20 by SciresM