Atmosphere-NX/Atmosphere #360: ERROR FATAL ERROR : FAILED TO STAT STAGE2 (ATMOSPHERE/FUSEE-SECONDARY.BIN)!

omarrimawi, 2019-02-21:
We would like to use GitHub to keep track of problems/feature requests.  
If you have a question, please join the ReSwitched discord for help. 
- Discord link: https://discordapp.com/invite/DThbZ7z

SciresM, 2019-02-21:
Please use the correct issue template.

Also sounds like sd card corruption.

closed 2019-02-21 by SciresM

omarrimawi, 2019-02-21:
SORRY I AM NEW HERE , SHALL I REFORMAT MY SD CARD ? AND IN WHICH FORMAT TYPE ?

DavidBuchanan314, 2019-02-21:
There is a key to the left of the "A" key on your keyboard labelled "CapsLock", please press it once before continuing.

darkpickles, 2019-02-26:
I have the same problem:-(

aveao, 2019-02-26:
Extract the zip properly.

cristianmiranda, 2019-08-20:
@omarrimawi @darkpickles Maybe too late, but better late than never 😄 

The problem should be the SD card format. If you're using FAT32 as I'm doing, check the scheme to be **FDisk_partition_scheme** and not **GUID_partition_scheme**

* _Green:_ correct
* _Red:_ wrong

[image]

Those are two 128GB cards with different format. If you're using macOS you can use the following command:

`sudo diskutil eraseDisk FAT32 SWITCH MBRFormat /dev/disk3`

(replace /dev/disk3 with your SD, you can check it with `sudo diskutil list`)

Hope it helps! 🙏

Conan179, 2019-08-20:
You are saying, switch donst load a fat32 from a gpt fort. SD card?

SciresM, 2019-08-20:
Please don't necro issues that have been resolved for half a year; it's bad etiquette.