Atmosphere-NX/Atmosphere #361: can't boot into CFW with 0.8.4

BaZZtian, 2019-02-21:
## Bug Report

### What's the issue you encountered?

### How can the issue be reproduced?

Everytime i load the 0.8.4 the Error shows up

### Crash Report

https://gist.github.com/BaZZtian/1965ea2bcffefaee426a815e53031ca5

### System Firmware Version

6.2.0

### Environment?

### Additional context?

BaZZtian, 2019-02-21:
nvm delete latest titel folder on SD fixed it ...

BaZZtian, 2019-02-21:
#close

closed 2019-02-21 by SciresM

powmaya, 2019-02-21:
Weren't you the one who was always playing, datamining and leaking Omega Ruby (and other Pokemon games) on 3ds ahead of release? If that's not piracy I'm not sure what is friend. 

Glass houses and all that.

https://projectpokemon.org/index/news/ultra-sun-and-ultra-moon-datamining-r66/
http://227.149.197.104.bc.googleusercontent.com/2016/11/08/pokemon-sun-and-moon-is-leaked-datamined/

bakugo, 2019-02-21:
@powmaya what exactly does this have to do with the issue submitted?

powmaya, 2019-02-22:
I posted it on the wrong one, thanks github.