Atmosphere-NX/Atmosphere #368: HDCP seems broken ( connected screen goes blank when opening hdcp enabled app)

fennectech, 2019-02-22:
## Bug Report

### What's the issue you encountered?

[ When opening youtube the screen goes blank and returns after going to home menu]</br>
[ no ]</br>
[ NA ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ 1. Simply open youtube app and screen goes blank. ]</br>

### Crash Report

[NA ]</br>

### System Firmware Version

7.0.1</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
fusee-primary.bin chainloaded with hekate.
- Official release or unofficial build: 
  - [ Official 0.8.4 ]
- Do you have additional kips or sysmodules you're loading:
SysFTPD
sys-clk 
- Homebrew software installed: [ Sys-clk applied to home menu but not the application.  This did not occor before 0.8.4 and 7.0.1 ]

### Additional context?

- Additional info about your environment:
- [ The issue only occored when i updated atmosphere  all settings and things previously existing with sys-clk and sys-ftpd worked fine in atmosphere 0.8.3]
 I hope i have done the thingy right this time.

SciresM, 2019-02-22:
Have you verified the issue occurs on stock 0.8.4 release without additional modules? Please always verify that.

fennectech, 2019-02-22:
I have just confirmed it happens on stock 0.8.4.

SciresM, 2019-02-22:
Does the issue occur in handheld mode, or only when docked?

SciresM, 2019-02-22:
Similarly: does the issue occur on 0.8.4 and downgraded-to-6.1.0?

fennectech, 2019-02-22:
Only in docked mode.  I’d rather not downgrade. Sorry.    Can you reproduce it on your end?

SciresM, 2019-02-22:
I do not own a television (by choice), and so cannot test anything involving docked mode.

My development switch also has a banned certificate, so I can't really download youtube...

SciresM, 2019-02-22:
(Sorry, meant to hit "comment", not "close and comment")

fennectech, 2019-02-22:
There is a modded YouTube that patches out the need for Nintendo network.

SciresM, 2019-02-22:
That does not solve the fact that I do not have a way of obtaining youtube, as I can't connect to the eshop.

fennectech, 2019-02-22:
Iirc its an NSP.

SciresM, 2019-02-22:
Atmosphere doesn't have signature patches, and I can't/don't provide support for warez.

fennectech, 2019-02-22:
Ill ask about for somebody to test :)

SciresM, 2019-02-22:
Could you also verify whether issue is specific to 7.0.1, or also affects 7.0.0?

fennectech, 2019-02-22:
Will do.

nebulakl, 2019-02-22:
Can confirm 7.0.0 is not affected.

fincs, 2019-02-22:
Cannot reproduce. My console is on 7.0.1 & docked & running AMS 0.8.4 without any extra kips or any other modifications. YouTube app works and I can play videos.

WinterMute, 2019-02-22:
Can't reproduce here either. 7.0.1 with ams 0.8.4, no kips or other mods, booted using latest fusee-primary via cortex M4 based dongle

fennectech, 2019-02-22:
Can comfirm 6.2.0 is not affected   (had some users try on discord.

fennectech, 2019-02-22:
Hmm  on 7.0.1 bone stock booted from tegrarcmgui  it flashes the HDMI icon a bunch of times (like its revebing signal and then not receving it) then eventualy starts working after a few minutes.   This is wierder.  Perhaps its my TV?  But everything worked before.

SciresM, 2019-02-23:
Sounds like a hardware problem.

closed 2019-02-23 by SciresM

fennectech, 2019-02-23:
ill downgrade