Atmosphere-NX/Atmosphere #369: Reboot to payload after SD removed (instead of power off)

Strat00s, 2019-02-22:
### What feature are you suggesting?
#### Overview:
Replace power off dialog option when SD card is removed with a reboot. Preferably with a reboot to payload.

### Why would this feature be useful?
Because everyone who removes the SD card is going to boot the Switch again anyways, so why do it manually when it can be automatic? ;)

SciresM, 2019-02-22:
Can't really reboot to payload if can't read the payload off the SD card :P

Strat00s, 2019-02-22:
> Can't really reboot to payload if can't read the payload off the SD card :P

You can always put it back in and then reboot. That's what I do now. But I have to go through the reboot menu.
So you can reinsert SD with whatever you put on it and then it would work as reboot to payload.

SciresM, 2019-02-22:
but...why are you removing the sd card?

Strat00s, 2019-02-22:
> but...why are you removing the sd card?

To add stuff to it? To add homebrew or whatever.

SciresM, 2019-02-22:
Since you have to reboot anyway...considered rebooting to memloader to access SD over usb?

Strat00s, 2019-02-22:
> Since you have to reboot anyway...considered rebooting to memloader to access SD over usb?

But then I have to reboot once again to RCM and push atmo again to boot back right?
I don't know. Seems really convenient, because most people still remove their SD to add stuff to it and this way you can remove it, put stuff on it, place it back and reboot back to CFW.
Or am I just missing something..?

SciresM, 2019-02-22:
It's more that atmosphere lacks any mechanism for modifying those menus atm, so changing a powerdown like that would impact all powerdowns...

hippydave, 2019-02-22:
Some people reboot to hekate payload - you can remove the sd, put stuff on it and put it back in while in hekate's menu. You can configure hekate to chainload fusee-primary to get you back into atmosphere.

ThatNerdyPikachu, 2019-02-22:
Couldn’t we load the payload on-boot into memory?

illzbee187, 2019-02-22:
You can run reboot to payload. When it reboots and says welcome atmosphere fusee remove the SD card so it errors before booting payload2. 
You will see 

'[ERROR] uSD: Failed to configure controller!
[ERROR] uSD: Failed to initialize the SDMMC driver!
[ERROR] Fatal error: Failed to mount at least on parition: I/O error

 Press power to reboot.'

Do what you need on the sd card and then reinsert press power to reboot it will boot the payload off the SD card.

linktlh, 2019-02-24:
I don't think this is really something **Atmosphére** should deal with. As @hippydave mentioned; instead you could replace the reboot payload with something like [Argon-NX](https://github.com/Guillem96/argon-nx) or [Hekate](https://github.com/CTCaer/hekate). At that point, once you reboot, you'd just boot straight into that other payload instead of the default atmosphere CFW firm. Once in the other payload, you could just eject the SD card 
<h6>I still wouldn't recommend ejecting the SD card without unmounting it. I know most people would say they've never had anything happen by doing that. While it may be a low risk, it's still a risk that you could corrupt your SD card.</h6>

SciresM, 2019-02-24:
Tentatively closing due to discussed issues. Will continue to think on it.

closed 2019-02-24 by SciresM

urherenow, 2020-04-27:
I know this is issue is old and closed, but I'm having sort of related trouble. I tried renaming both the latest hekate.bin and even nyx.bin ro reboot_payload.bin, and replacing it on my card (without rebooting, using ftpd). Rebooting still goes through the sept screen and loads right back into Atmosphere. How? cold-boot with the same hekate.bin gives me the nyx menu with my current configuration.

Genwald, 2020-04-27:
@urherenow That is the expected and intended behavior afaik. Atmosphere includes a homebrew that can be used to reboot to a new payload if the payload has changed after boot.